EdgeTX/edgetx #4357: [X12s] MPM module connectivity issue

Unb0rn, 2023-11-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a modded X12s device (ISRM, externalaccessmod and so on), I've had some problems with my multiprotocol module where I got only NO_TELEMETRY message. (Described it [here](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/621#issuecomment-1741702661))
After upgrading to 2.9.2 the issue has been kinda resolved - I can now make MPM work but it involves some strange actions. By default I still get this no telemetry message but then if I change external module in model settings to something else and back, the module gets detected, I can run spectrum scan on external module without any stutters and even flash it successfully. The module's fw version is displayed correctly in system info. This workaround works until reboot.

One more thing - this switching back and forth works with some options and doesn't work with others - like selecting Flysky and then back Multi module works fine, but selecting R9M and back Multi still produces no telemetry.
I have no idea how to debug it further...

### Expected Behavior

MPM module gets discovered every time when it's selected in model settings.

### Steps To Reproduce

1. Select MULTI in external RF
2. Reboot
3. Get No telemetry
4. Select Flysky
5. Select MULTI back
6. IT WORKS!

### Version

2.9.2

### Transmitter

FrSky X12

pfeerick, 2023-11-23:
Can you backup your RADIO and MODEL folders, and try with a nightly build? There were some changes around use of the SPort lines which could be relevant to you.

Unb0rn, 2023-11-24:
> Can you backup your RADIO and MODEL folders, and try with a nightly build? There were some changes around use of the SPort lines which could be relevant to you.

Actually it got worse!) Looks like it's somehow tied to this FLYSKY external module mode. This newer nightly build - cdfe7b58 has no FLYSKY option. 
By default MULTI has this "NO TELEMETRY" message and then selecting any other option first and then back to MULTI results in "no serial data" message. 

Flashing back 2.9.2 and running through the steps I described previously returns MPM back to normal.

One more thing - in m X12s unit I have inverter replaced with the faster one (CRSF/ELRS work fine and R9M/ACCESS too), maybe it has something to do with it?

ParkerEde, 2023-11-25:
Today I have ETX2.9.2 and my Horus X10S Express with an IRange IRX+ (MPM) connected to a Blade130S and suddenly lost all control.  This has happened twice in a short space of time. Luckily nothing happened to the Blade 130S, it happily fell out of the air. The red control LED then flashed on the Irange IRX+ module. Normally it lights up permanently. I then flashed back to ETX 2.9.1 and the problem did not occur again. What happened with 2.9.2 that the serial connection to the MPM can no longer be kept stable?

ParkerEde, 2023-12-10:
today I got the same problem again. There is definitely still a problem in the serial control of the MPM.

Unb0rn, 2023-12-15:
@pfeerick Regarding the config - exactly, aux1 is dedicated to ACCESS (External module in settings), Sample mode is set to "Normal" (as I said, the inverter is replaced and I have no problems with any ELRS modules on any rate, just for example)

I tried again on the latest nightly (42e245b1) - the result is the same. Setting the module to AFHDS3 and back to MULTI still proucest the old "no serial input", rolling back to 2.9.2 and doing the same thing - setting external module to FLYSKY and back to MULTI results in the correct version detected, external spectrum scan working without any hiccups and so on.
And of course, Multi module still doesn't work for me without these manipulations mentioned above - neither for 2.9.2, nor for nightly.

Feels like we need to add some reset or something.

richardclli, 2023-12-18:
Flysky external module is FRM303, and it works only with AUX1 and radio settings set AUX1 to external modules in X10 based radios.

I am not sure what you have modded in your X12S, if you can give me more details, I will see if  I can help.

> > Can you backup your RADIO and MODEL folders, and try with a nightly build? There were some changes around use of the SPort lines which could be relevant to you.
> 
> Actually it got worse!) Looks like it's somehow tied to this FLYSKY external module mode. This newer nightly build - [cdfe7b5](https://github.com/EdgeTX/edgetx/commit/cdfe7b588af3655f7bed0158551b5d6643ba64ba) has no FLYSKY option. By default MULTI has this "NO TELEMETRY" message and then selecting any other option first and then back to MULTI results in "no serial data" message.
> 
> Flashing back 2.9.2 and running through the steps I described previously returns MPM back to normal.
> 
> One more thing - in m X12s unit I have inverter replaced with the faster one (CRSF/ELRS work fine and R9M/ACCESS too), maybe it has something to do with it?

Unb0rn, 2023-12-18:
@richardclli Thanks! I'm ready to answer any questions.
Regarding the radio itself - it is an ordinary X12s(not the factory ISRM version), everything I replaced/modded myself.
List of X12s mods I use:

1. XJT radio module replaced with ISRM (the same upgrade FrSky sells for X10s)
2. Bluetooth module replaced to a new version from the same upgrade bundle (the new PARA thing - I think this is the only mod I don't actually use.
3. Stock "slow" inverter has been replaced to a faster one like described [here](https://blog.seidel-philipp.de/fixed-inverter-mod-for-tbs-crossfire-and-frsky-qx7/)
4. External ACCESS mod applied to handle my R9M 2019
5. Replaced the stock screen with an I2C touchscreen

The MPM (IRX4+) I use has been modded too as the author suggests [here](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/621#issuecomment-932708948)

Everything (except for MPM) seems to work fine - I train from time to time using XSR-SIM, so ISRM works fine, never had any problems with it. I've also tried flying with ELRS (but for now it's only my DIY 433 version, waiting for hi-speed 2.4 receivers to arrive, so actually not a good test for inverter; however ELRS app works fine with any baudrate, it doesn't hang and doesn't show signs of high communications error rate), even the MPM seems to work fine (tried indoors with a whoop, not for long obviously, as well as Lua Spectrum Scan EXT) but it still requires this switching to FLYSKY back and forth on every reboot on 2.9.2 and doesn't work at all in nightlies.

I haven't tried long-range with R9M in ACCESS for a long time, almost lost my interest in it, but I can say the module is identified and reports the correct version info.

Unb0rn, 2024-08-04:
After updating to the latest 2.10.3 I can confirm that the problem still exists.
The behavior now is:

1. Internal ISRM + External CRSF - Doesn't work at all
2. External CRSF Only - Works!
Switching between port speeds has issues:
- Setting baud rate to 400k results in `Status: 1000 Hz` and nothing works, even elrsv3 lua script is unable to communicate with the module, but this problem can be "fixed" by switching the internal ISRM on and back off (it really feels like this action resets the bus or something). After that 400k produces the correct `Status: 500 Hz` and elrsv3 lua is back to normal
- Setting baud rate to 115k produces `Status: 150 Hz` and everything is alright
- Setting baud rate to 921k is exactly the same as 400k, but it cannot be "fixed" the same way as 400k - flicking ISRM on and off still results in `Status 1000 Hz` 
- Setting baud rate to 1.87M resolves all the issues - Axis Thor 2400 runs at full 1000Hz, no errors in elrsv3 lua 
3. R9M ACCESS - seems to work fine with both ISRM on and off (probably due to using a separate aux1 UART)
4. Internal ISRM + MSM - Doesn't work at all (`No MULTI_TELEMETRY` no matter what I switch)
5. MPM Only - Doesn't work
After reboot it produces `No MULTI_TELEMETRY`,  but switching ISRM on and off OR switching to some other mode - AFHDS3, for example turns the status to "No serial input" and cycling the protocols inside MPM (for example Flysky -> FrSky D) occasionally shows the correct MPM version and then back to "`No serial input`"

Additional note: this internal ISRM flicking seems to be more "powerful" than reboot - if the system boots with CRSF 400k it still produces this incorrect status and module connectivity loss, but flicking ISRM on and off "resolves" the problem
Maybe the system selects incorrect communication speed? Like 400kbaud port is unable to communicate full 1000Hz, but it still tries for some reason.

It feels like either something is being incorrectly reset or wrong port speed is chosen.
Can we try to limit the port speed? Is there any way to do it for MPM, maybe through the console?

TheFuryX, 2026-09-19:
My Tx12mk2 does the same thing with iRange MPM that i "extended" into the plus version. My solution for this bug seems to be whenever the issue occures, i reflash the module via radio menu and then it works for a few restarts or protocol changes. 
The MPM module i have is some cheap ass one from Ali and it uses APM32 (allegedly already manufactured by Geehy not apexmic) micro instead of the one made by STM - but i believe its fake to some extent anyways, logo is marked at different time, using different machine/technique and has almost no depth while being out of focus. On the other hand the rest of the symbols on the package are done later on, crisp and deeper. The "arm" stamp actualy seems to be the same age as the geehy logo. 
My guess would be the problem is either with the actual flash size (corruprion while indexing avalible protocols - i used to have no problem with it until i dumped a shitload of protocols into the poor thing)... or the voltage regulator cant keep up, which again can fluctuate while accessing second half of the flash... for some reason i trust its sram to be stable but i digress 💁🏻‍♂️
So... try to flash the mpm and see if the problem dissappears for a while in your situation

TheFuryX, 2026-09-19:
[image]

[image]

Here is a sneak peek inside my MPM

[image]

And here older variant of said uC branded by the original manufacturer before geehy took over. That one i actualy believe is genuine. Oh and btw apm32 is supposed to run native at higher clock compared to STMs design while being ok running underclocked for compatibility reasons. Nvm the fact its native usb works which is kinda strange given 72:2 != 48 right? :P (the reason f103 is not officialy usb capable)

Nvm, it has 1/**1.5**/2 avalible prescalers for the usbdclk, so its capable of running usb up to spec on 72mhz, but im using precompiled binary for stm so hopefully the difference of 3 prescalers avalible vs 2 inside stm is not the issue 🤔