EdgeTX/edgetx #4358: TX16s MK I with EdgeTX 2.9.2 "Providence" Crashes on Joystick mode when pluged thru type C cable

omersal150, 2023-11-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

I plug in the remote and play on realflight evolution and the controller just crashes and enters emergency mode.
It happened to me on edgetx 2.9.0 and 2.9.1 and nothing helps it.
I have a 32 GB fat32 sd card installed correctly i tried other cables with the same result.
Tried removing widgets like other issues reported on earlier versions and nothing helps it just keeps crashing.
My battery is good i tried both 2 18625 batteries with the connecter it came with and the original 5000 mah 7.4 lion battery for tx16s. i have a new original SD card, everything on the controller is good but it keeps doing it no matter the simulator, and can happen at any time from a few seconds to a crash or even 10-15 min.SD

### Expected Behavior

Not Crash every few min =

### Steps To Reproduce

1. Plug in Controller
2. Choose Joystick Mode
3. play a bit on a simulator
4. Error happen

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

omersal150, 2023-11-25:
I found out that if i go to any menu like model settings or system the issue is gone so for now i just stay and model settings and it works great.

Mybe the issue is with the widgets? I only have sticks menu to see the channel outputs

3djc, 2023-11-25:
This seems like a duplicate of #4262

philmoz, 2023-11-25:
> Mybe the issue is with the widgets? I only have sticks menu to see the channel outputs

Do you mean the 'Outputs' widget?

Are there any widgets in the top bar?