EdgeTX/edgetx #4361: Some Mixes no longer work in v2.9.2

claudevandaele, 2023-11-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When updating from v2.9.1 to v2.9.2, I notice that several mixes no longer fully work under v2.9.2. . I have a rescue mode that skips a line and a spinfix mix that no longer works.
When reinstalling back to 9.1, everything works normally again. Anyone else having such problems? 

           
[image]
                                                                                                                                                                                 

[image]

[image]

### Expected Behavior

Mixes work in v.2.9.1 not in v.2.9.2 !

### Steps To Reproduce

1. Switch SH
2. R0 L04 R0 active (9.1) -> 9.2 not working
3. R1 L05 R1 active
4. ....

1.Switch SG or SF activate SpinFix (9.1) -> 9.2 not working 

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

philmoz, 2023-11-25:
Please upload a copy of the affected model file and your radio.yml file.

claudevandaele, 2023-11-26:
[MODELS.zip](https://github.com/EdgeTX/edgetx/files/13466920/MODELS.zip)

claudevandaele, 2023-11-26:
Model and Radio file in comment.

claudevandaele, 2023-11-27:
Solution in v.2.9.3 ?

philmoz, 2023-11-27:
> Solution in v.2.9.3 ?

Hopefully. The PR needs to be reviewed and tested.

claudevandaele, 2023-11-27:
Thanks for the reply, and maybe a quick solution. Grz. Claude

ParkerEde, 2023-11-29:
I've tested the PR. It's okay.

PR #4364 (2023-12-21): fix(radio): Mix delay not working.
Fixes #4361

The changes in PR 4256 to fix issue 4254 did not quite work as expected and mix delays stopped working.

Changes:
- Reworked the logic so delays work as they did prior to 4256.
- Renamed variables and structs for readability
- Disabled invalid test 'DelayOnSwitch2' (SWSRC_ON is not a valid switch value for a mix line).

Also tested that the issue raised in 4254 is still fixed.
closed 2023-12-21 by pfeerick