EdgeTX/edgetx #4363: SSL Handshake Failed Error in EdgeTX Companion 2.9.2 on Linux Mint 21.2

neiltarar, 2023-11-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

**Description:**
When attempting to use the "Download components" feature in EdgeTX Companion 2.9.2 on Linux Mint, I am encountering an SSL handshake error that prevents the download of firmware updates or other components. This suggests a potential issue with SSL bindings within the AppImage that did not exist in version 2.8.5, where the issue was previously reported and resolved.

**Relevant Discussion:**
This issue was previously reported and resolved in discussion [#3593](https://github.com/EdgeTX/edgetx/issues/3593)

**Additional Details:**
Upon extracting the AppImage contents for version 2.9.1, libssl.so and libssl.1.1 are present within the lib directory, indicating proper SSL support. However, in the 2.9.2 version, these files are notably absent, which may correlate with the SSL handshake failures currently being experienced.

[image]

### Expected Behavior

The EdgeTX Companion should search for and prompt updates, allowing the user to download necessary components without encountering SSL errors.

### Steps To Reproduce

1. Download the edgetx-cpnlinuxv2.9.2.zip file from  https://github.com/EdgeTX/edgetx/releases/tag/v2.9.2
2. Extract and set the AppImage within as executable.
3. Navigate to Settings > Settings > Update Settings in EdgeTX Companion, ensuring that all options under "Check Release channel" are selected.
4. Click on the "Download components" icon in the Companion UI.

### Version

2.9.2

### Transmitter

RadioMaster Boxer

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.2

closed 2023-11-26 by neiltarar

neiltarar, 2023-11-26:
I found that the same issue was mentioned in another report

https://github.com/EdgeTX/edgetx/issues/4316

dlabrie55, 2023-12-14:
I am having the same problem on Linux, Cinnamon. 
[image]

neiltarar, 2023-12-17:
@dlabrie55  Did you try manually installing libssl1.1 from [http://archive.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1f-1ubuntu2.20_amd64.deb](https://github.com/EdgeTX/edgetx/issues/url)

I wonder if it will be added to the AppImage dependency list with the next update...

dlabrie55, 2023-12-17:
No, I haven't. I will give it a try.

David LaBrie
Campton, NH

dlabrie55, 2023-12-18:
Having added
http://archive.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1f-1ubuntu2.20_amd64.deb
<https://github.com/EdgeTX/edgetx/issues/url>

I was able to download everything.
Thanks.

David LaBrie
Campton, NH