EdgeTX/edgetx #4365: TX16S enters Emergency Mode when swiching to Special Functions page

Kai2510, 2023-11-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

TX16S enters Emergency Mode when swiching to Special Functions page. It occurs every time, but model data is saved. 
The model setting is copied from a setting file of x9d+, which includes a Vario function for gliders.

### Expected Behavior

The transmitter is able to change SF settings. 

### Steps To Reproduce

1. Enter the model setting page.
2. Switch to Special Function page via touch screen.
3. TX16S enters Emergency Mode.

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

Kai2510, 2023-11-28:
> Please upload the affected model file and the radio.yml file from your SD card.

Please refer to this configuration file. The model file is "gomini".
[zzf-tx16s-231126-gomini.zip](https://github.com/EdgeTX/edgetx/files/13488478/zzf-tx16s-231126-gomini.zip)

philmoz, 2023-11-28:
The model does not crash on my TX16S or in the simulator.

Please copy the files directly from the SD card - reading them in Companion and saving as an ETX file probably removes any incompatibilities.

Dronitoni, 2023-12-01:
I see the same with my new radio TX16s. Emergeny mode is showing up when I tap on the main screen and select the "manage models" icon. It's repeatable, I tried it several times.
Version 2.9.2
I flashed this file: tx16s-8faf131.bin

pfeerick, 2023-12-03:
@Dronitoni You should probably open a new issue, as I believe you issue to be differeent (since yours is related to the Manage Models screen, not the Special Functions screen). And most likely something related to one of the model images, or possibly a model name.

Dronitoni, 2023-12-04:
@philmoz thank you.

Have you ever seen Emergency Mode during a flight?

Kai2510, 2023-12-04:
> The model does not crash on my TX16S or in the simulator.
> 
> Please copy the files directly from the SD card - reading them in Companion and saving as an ETX file probably removes any incompatibilities.

Please refer to these files. In my mind Model18.yml is "GOMINI", where the bug was triggered.

By the way, I also encountered a problem: the model list displayed on the radio was incomplete after I flashed the configuration by Companion. I don't know if it's a bug or a feature.

[MODELS.zip](https://github.com/EdgeTX/edgetx/files/13543460/MODELS.zip)
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/13543464/RADIO.zip)

Dronitoni, 2023-12-05:
My problem is solved. (hopefully).

The Radiomaster guys pointed me into the direction of a potentially faulty SD-Card.
I replaced the original SD-Card with an old one I had.
The result is, that the error I had was gone, at least for the last 2 days.

I formatted a 32GB-Card to 4GB in FAT32 and 4KB-Blocksize as described here:
https://youtu.be/aNBJzDKAkog?t=2156

pfeerick, 2024-03-04:
> Please refer to these files. In my mind Model18.yml is "GOMINI", where the bug was triggered.
>
> By the way, I also encountered a problem: the model list displayed on the radio was incomplete after I [...]

@philmoz Would you mind following up on this :smile:

philmoz, 2024-03-04:
> @philmoz Would you mind following up on this 😄

Not reproducible by me.