EdgeTX/edgetx #4371: Widget APP MODE

JimB40, 2023-11-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Since double tap is not coming anytime soon, using LUA apps (Yappu, iNav, Agent Lite, ELRS config, Toolbox etc) )that works as widget in full screen mode became cumbersome.
When switching to the Main View with LUA app to use key inputs user needs to:
1. Long press ENTER
2. Select Full Screen
3. Shot press ENTER

Switching between screens is quite often task so this makes UX dauting 

### Describe the solution you'd like

Special mode in widgets layout called "APP" 

1. It is full screen always.
2. No Main view options (Top Bar, Flight Mode etc) or those option are set to inactive when APP MODE is selected for widget 
LUA App working as widget in Full Screen is special case
3. No widget frame is selected and no contextual menu with Full Screen and Widgets Options
- no need for widget selection frame as there is only one wdget without any system UI elemnts.  
- no need for Widget Oprionrs - biger apps do not use system options as ussually they have their own configuration system)

How it works:
1. User selects Main View with LUA APP working in APP MODE - key inputs for LUA APP are inactive 
2. Long press ENTER - activates key inputs for LUA APP. (insetad of showing widget selection frame)
3. Long press EXIT - disable key inputs for LUA APP returning UI to system

This will allow to navigate efficiently between Main Views with LUA APPS working as full screen widget and other Main Views.

philmoz, 2024-02-16:
I've read through this and PR 2629 and I think I understand the issue - I agree the current system is very cumbersome.

What I would suggest is:
- Add an 'App mode' toggle to the view setup page. This would not need a separate setting; but is just a shortcut to turn off the topbar, trims, sliders and flight modes.
- The default 'app mode' view would be the selected widget (full screen); but also still showing the EdgeTX button in the top left corner (visual indicator of mode). Keys and touch events are not passed to the widget. Swiping can be used to change view and enter/touch brings up quick menu.
- A long ENTER press or long touch on the screen hides the EdgeTX button and directs key and touch events to the widget. The user will not be able to swipe to change views, or open the quick menu.
- A long press of RTN key shows the EdgeTX button and stops events/touch going to the widget.

In the case where the main view has multiple widgets, topbar, sliders etc, then the current long press on the widget would still open the popup menu; but with the 'Fullscreen' option renamed to 'App mode'. Selecting this would put the widget in the same app mode (full screen, no EdgeTX button and inputs going to the widget). Long press RTN restores the previous main view.

This would allow us to get rid of the 'standalone' Lua mode in the code and should simplify the way widgets are handled.

Looking at the code I think this will be easier to do using PR 4407 as a base. Unless someone else wants to take it up I will have a go at it for 2.11.

philmoz, 2024-02-16:
Screen shot example:
[image]

JimB40, 2024-02-17:
Well what can I say @philmoz, You grasped concept 100% :)

1. I'd rather set this controls off and hide them. Less oportunities that user will switch them on anyway  

2. Visual confirmation which state is active I'd consider Bubble msg you've coded, so UI communication is consistent
Shouldn't  too big so It can say something like:
"Display only mode"
- When user selects main view with appMode layout -> overlay with bubble msg is constantly visible
- With long ENTER press bubble msg hides
- With long RTN press bubble msg is visible again

pfeerick, 2024-02-17:
Please also consider #825 when thinking of this, as rather than it being a single fullscreen layout, it would nice if this applied to any widget layout.  But is by no means a requirement.

philmoz, 2024-02-17:
@JimB40 I've made an initial PR #4655 for this.

Please give it a try and see what you think.

It uses the 'EdgeTX' button in the top left to differentiate mode - at the moment I need this to have a way to open the quick menu (tapping on the widget will not work).

'App mode' is enabled by creating a layout with 1 widget (the 'full screen' layout) and turning off the topbar, flight modes, sliders and trims.

It is based on 4407 so may have bugs related to that one included 😄

PR #4655 (2024-05-30): feat(color): widget app mode
Fixes #4371 & #2629 

Rework the way 'full screen' mode is selected for widgets.

> It uses the 'EdgeTX' button in the top left to differentiate mode - at the moment I need this to have a way to open the quick menu (tapping on the widget will not work).
>
> 'App mode' is enabled by creating a layout with 1 widget (the 'full screen' layout) and turning off the topbar, flight modes, sliders and trims.
>
> - The default 'app mode' view would be the selected widget (full screen); but also still showing the EdgeTX button in the top left corner (visual indicator of mode) [... cut]
closed 2024-05-30 by pfeerick