EdgeTX/edgetx #4374: Discovered telemetry sensors lost after power of

krzysztofkuczek, 2023-12-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After discovering CRSF sensor, everything seems to be ok, but after powering the radio off, and then powering it on again, the list of sensors is empty.

### Expected Behavior

The list of sensors should be the same after power cycle.

### Steps To Reproduce

1). Go to telemetry tab
2). Discover sensors
3). Stop discovering
4). Power cycle
5). Got o telemetry tab, the sensors list is empty,  but it should not be empty

The all available firmware version are affected including n build from 2.12.2023

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster Pocket

PYJTERSTM32, 2023-12-02:
I have the same problem, after restarting I have to search for sensors again

pfeerick, 2023-12-21:
Is the list empty, or is it simply that you need to un-collapse the list? If this arrow is pointing down, you need to press enter while it is highlighted (like it is in the screenshot) to un-collapse/expand the telemetry sensors list...

Collapsed:
[image]

Expanded:
[image]

krzysztofkuczek, 2023-12-22:
> Is the list empty, or is it simply that you need to un-collapse the list? If this arrow is pointing down, you need to press enter while it is highlighted (like it is in the screenshot) to [...]

Yes, you are right. The list is simple not expanded, and maybe it should be ;). Anyway it does not required to discover sensors every time, so it's quite ok.
Thank you
Kris