EdgeTX/edgetx #4379: LUA: custom script returns wrong values

mha1, 2023-12-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Simple custom script returns correct value in value widget but wrong value in custom scripts screen and announcing wrong value with play value.

### Expected Behavior

consistent use of calculated custom script value

### Steps To Reproduce

1. put simple custom script test1.lua in folder SCRPITS/MIXES:
```
local output = { "val1" }

local function run()
  local val1 = 2724
	
  return val1
end

return { run=run, output=output }
```
2. go to model settings / custom  scripts, select test1 and name it tst1
3. observe tst/val1 2660 (expected 2724)

[image]

4. setup SF play value to announce value of tst1/val1
5. activate SF and observe announced value 2660 (expected 2724)

[image]

6. setup value widget on main screen to show value of tst/val1
7. observe value shown 2724 (expected 2724)

[image]

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

[image]

philmoz, 2023-12-05:
The value shown on the Custom Scripts edit page ('Outputs'), and the value used by the 'Play Value' SF are scaled:
    scaledValue = (scriptValue * 1000) / 1024

The 'Value' widget shows the raw value returned from the script.

I'm guessing the intention was to scale the script output to a % value (x10), and it was assumed that the script output would be in the range -1024 to 1024.

Simple fix would be to apply the scaling logic to the Value widget.

mha1, 2023-12-07:
@pfeerick your take
@3djc I'm sure you an opinion too

3djc, 2023-12-07:
As always, don't break existing scripts...

philmoz, 2023-12-07:
Maybe the custom scripts need a 'units' property, like GVars, to control how the output values are interpreted.

mha1, 2023-12-07:
or just an option in the custom scripts menu:
script type: mixer source or custom

type mixer source as default to keep @3djc happy for not breaking existing scripts with output scaling and type custom to output raw unscaled values

mha1, 2023-12-07:
> Maybe the custom scripts need a 'units' property, like GVars, to control how the output values are interpreted.

Thinking again: Great idea. Allow to individually assign units for custom script outputs plus defining a new unit "MixSrc" would solve the problem in an elegant way and even add additional value to custom scripts. "MixSrc" as default for backwards compatibility.

PR #4391 (2024-07-25): fix: inconsistency between SD play value, model custom script screen and value widget
Fixes the custom script output inconsistencies between SD play value, model custom script screen and value widget. This still allows the use of Custom Scripts as Mixer Source but also in a general way to calculate arbitrary data that can be used in widgets, voice messages and LUA scripts. Backwards compatibility for mixer scripts is maintained. 

Fixes #4379 

Summary of changes:
- SF play value now announces script outputs as raw value
- model custom script screen now shows script output raw values

Unchange [... cut]
closed 2024-07-25 by pfeerick