EdgeTX/edgetx #4385: [Zorro] On/Off Button blue led flashing after shutdown

TheIsotopes, 2023-12-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Since #3846 the on/off button flashing blue and and the speaker crackles after shutdown.
Just removing the battery ensures peace of mind.

Imo there seems to be a problem with the bootloader. @raphaelcoeffic pls can you check this.

### Expected Behavior

fix the current behavior

### Steps To Reproduce

flash latest master and shutdown 

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster Zorro

pfeerick, 2023-12-10:
I'm not seeing this on Zorro with internal MPM or ELRS with the latest nightly. Is there anything else needed to trigger this? i.e. has the TX been modded for the RTC brownout issue, is a external module in use, is it getting externally powered, is it only happening if the RF link is active and you force it to shut down, etc?

TheIsotopes, 2023-12-10:
The TX is modded with an extra capacitor for the RTC brownout issue and is not connected to an RX.

TheIsotopes, 2023-12-10:
I removed the cap and it shutdown normally again.

But I noticed another problem, i still use the last nightly ecd50cf :
When I press `SYS` and then `PAGE>` the screen briefly goes off and then on again, the battery indicator briefly shows 0.0V and an exclamation mark flashes in the top left corner.

3djc, 2023-12-10:
>  exclamation mark flashes in the top left corner.

That means emergency mode. Bad....

pfeerick, 2023-12-10:
That I can reproduce, and need to check if it's specific to the zorro. It is pretty consistent though - power on - `SYS` - `PAGE>` - EM. If you `PAGE<` first, it *may* not happen when you then `PAGE>` twice, but more likely than not it will.

TheIsotopes, 2023-12-18:
Any news about this issue?
I also had to notice that the zorro does not restart after a bootloader exit.

TheIsotopes, 2023-12-19:
@raphaelcoeffic @pfeerick So far, so good... the problem with the bootloader exit is fixed in #4434.

But there is still the problem with the EM :-(

pfeerick, 2023-12-19:
#4436 should resolve that

TheIsotopes, 2023-12-20:
everything fixed, I think we can close here.

closed 2023-12-20 by TheIsotopes