EdgeTX/edgetx #4398: Rotary Encoder Reverse Direction

swissfreek, 2023-12-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It is currently possible to reverse the rotary encoder. However, currently if you turn the encoder one direction, it makes the selection move "down" in the menu system (for example, model selection), but turning the encoder the same direction makes number or letter inputs go "up" (for example, 0-1-2/A-B-C). For my brain this is unintuitive, as going "up" in the menu (bottom of screen to top of screen) should also make the numbers or letters go "up". Additionally, contextual menus follow the same logic as numbers, so on different menus one direction is "up" and the other is "down".

If you go in system settings and change the direction of the encoder, both issues reverse, so they move in the opposite direction, but still not synchronized.

### Describe the solution you'd like

One of two solutions:

1) Reverse the direction of the encoder in the main menu (model selection screen, mixer page, etc.) so that whatever direction of the encoder makes the cursor scroll from top of screen to bottom is the same direction as what makes the value increase when inputting numbers or letters. The contextual menus are already this way.

2) Allow users the option of changing both scroll directions independently in the System settings. So one setting for menu scroll direction, one setting for number/letter entry scroll direction. To me the only logical way to do it is the way I describe in 1), but if other people disagree, then I would like the option.

For what it's worth, TeamBlacksheep's FreedomTX, also forked from OpenTX, implemented item 1) above for their radios. So the code already exists in GitHub for easy implementation, but I suspect it's a matter of adding a single minus sign somewhere in the menu code?

pfeerick, 2023-12-09:
What radio is this on? And what version of EdgeTX are you running? B&W radios introduced more options for the rotary encoder in 2.8 (https://github.com/EdgeTX/edgetx/pull/2045), but it's not been implemented on colorlcd as it's not so easy to implement.

swissfreek, 2023-12-09:
Sorry, yes, this is on an MT12, so B&W radio. Version says “RM Factory”, dated 28 October 2023.

pfeerick, 2023-12-10:
2.8 was released in Nov 2022 ;) You should have four options Normal, Inverted, V-I H-N, V-I H-A.  V-I H-N (vertical invert, horizontal normal) is probably what you want.

swissfreek, 2023-12-10:
Ahh, thanks. V-I H-N is the exact OPPOSITE of what I'm looking for. So I guess what I want is V-N H-I? But there is no V-N option.

pfeerick, 2023-12-10:
V-N H-N would be "Normal" ;) V-I H-I would be "Invert". So yes, you are after an option that doesn't exist (yet?). 

We'll have to see how things go with the MT12, as this is the first radio that has the rotary encoder horizontal to the display... all the other radios are basically vertically aligned. Another way to look at this is when you roll the encoder down, the numbers get bigger, and they get smaller when you roll the encoder up. To **_me_**, that makes more sense (the way it is) than the other way around, but I'm not averse to another option being added for the opposite.

swissfreek, 2023-12-10:
Yes, this is the way it is (or was) in Audis. To me this is backwards but I understand not everyone's brain works the same. And you are right, it's the first time the roller is horizontal which makes it a new perception. I'm hopeful that since all the framework is apparently there, an option could be added for V-N H-I.

3djc, 2023-12-10:
I'll take it

3djc, 2023-12-10:
@swissfreek , could you describe the behavior you want, I'm not sure I'm getting it right.

When moving your finger from radiomaster logo to the belt thing;
- what direction should menu move
- what direction should value move

pfeerick, 2023-12-10:
Menu unchanged (i.e. V-N)
Value opposite to what it is (i.e. H-I) 

So when changing things like numbers, they decrease when from logo to belt, rather than increase.

pfeerick, 2023-12-10:
https://github.com/EdgeTX/edgetx/pull/2045 should be handy as a reference to make sure nothing is missed ;)

swissfreek, 2023-12-10:
> #2045 should be handy as a reference to make sure nothing is missed ;)

Shoot, I didn’t search well enough, I’m sorry.

pfeerick, 2023-12-10:
@swissfreek No, that was for JC, since this was probably going to piggy back that. Give #4401 a try and see if that is what you were after. 

> To access the automatic builds for a PR, first open up the PR page. Under the PR title, you should see a series of tabs (Comments, Commits, Checks, Files Changed). Click on the Checks tab. Then if you are after the radio firmware, click on the link on the left side mentioning firmware. If you want a Companion/Simulator build, pick the one that suits your operating system. Then, under the 'Artifacts' heading near the bottom of the page will be a link to the zip file containing either the firmware or companion package.

swissfreek, 2023-12-10:
> @swissfreek No, that was for JC, since this was probably going to piggy back that. Give #4401 a try and see if that is what you were after.
> 

YES! Thank you, that's exactly what I'm looking for.

closed 2023-12-11 by pfeerick

swissfreek, 2023-12-11:
I realize I'm probably being a persnickety pain in the ass, but is it not possible for the contextual menus to match the rest of the scrolling lists?

pfeerick, 2023-12-11:
Yeah, I noticed that, looking into it.

chofchop, 2023-12-25:
My MT12 (nightly build mt12-fbe06b3.bin) can now be operated almost intuitively with the V-N E-I option.
It's a small detail, but
**Pwr On delay
Pwr off delay**
However, the operations and numerical changes are still going backwards.

PR #4401 (2023-12-11): feat(bw): Add inverted edit option for rotary encoder
Allow user(s ?) to invert rotary encoder direction for editing only

This fixes #4398