EdgeTX/edgetx #4412: Misinterpreted setting of Special Functions parameters Repeat and Enable

MomoBrut, 2023-12-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

If the Repeat and Enable parameters are somehow set on the Special Functions Companion Tab for the Radiomaster Pocket transmitter, these parameters will not be interpreted correctly after uploading the model to the transmitter. Specifically, all items of the Play Track type are always played repeatedly with an interval of 1s, regardless of how the Repeat and Enable parameters were set in Companion. These parameters must always be manually set to the correct values only in the transmitter, it is simply not possible through the Companion. The reverse process of uploading a model from the transmitter to the Companion works correctly, both Repeat and Enable parameter settings are transferred correctly.

### Expected Behavior

The parameters set in the Companion must always be correctly transferred to the transmitter.

### Steps To Reproduce

Anyone who knows how to use Companion knows how to verify this...

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster Pocket

### Operating System (OS)

Windows

### OS Version

10

### Anything else?

Nightly 2.10.0-main, Dec 6 2023, commit 7206182

philmoz, 2023-12-12:
Are you running 2.10 firmware on your radio?

3djc, 2023-12-12:
He must be, pocket is 2.10 based only

philmoz, 2023-12-12:
OK - maybe not an up to date 2.10 version on the radio?

It works ok on my TX16S.

3djc, 2023-12-12:
Any nightly should have one

MomoBrut, 2023-12-12:
Yes, I noticed. But I don't really want to put it on the production transmitter. If I understand correctly, the Companion is fine, the problem is non-standard firmware in the transmitter. It is so?

pfeerick, 2023-12-12:
You can use the nightly firmware for the Pocket. You can flash it via buddy https://http:buddy.edgetx.org/#/flash?version=nightly&target=pocket

You should be using the matching firmware and Companion. So nightly firmware and a recent build of Companion

> For the latest version testing version of Companion, from the link below for your preferred operating system, pick the most recent successful build (will have a green checkmark next to it). On the resulting page, the link to download the installer package is under the Artifacts heading near the bottom.

Companion (main branch): 
- Windows =>  https://github.com/EdgeTX/edgetx/actions/workflows/win_cpn-64.yml?query=branch%3Amain
- Mac => https://github.com/EdgeTX/edgetx/actions/workflows/macosx_cpn.yml?query=branch%3Amain
- Linux => https://github.com/EdgeTX/edgetx/actions/workflows/linux_cpn.yml?query=branch%3Amain

3djc, 2023-12-12:
And there is no such thing as non standard firmware here, but it could be because of the time gap between radio firmware and companion

MomoBrut, 2023-12-12:
I see it as not standard, its name is "RM Factory" (1fdb58ba), 2023-07-27

3djc, 2023-12-12:
Very standard. That firmware has been built by us using this code. Those firmware are then tested by manufacturers, who then install those at factory. Nothing special about 5hose, just more tested

MomoBrut, 2023-12-12:
OK, thank you for explanation, I will test nightly build later.

MomoBrut, 2023-12-13:
I uploaded nigthly build 9e114c4d to Pocket. The original problem is solved, Special functions work as they should. But another appeared. The Global Functions setting in the Radio Menus is not transferred correctly to the transmitter, even if the item is Enabled in the Companion, it is always Disabled after transferring to the transmitter. There may be more, but this is what I noticed right away.

MomoBrut, 2023-12-14:
No response?

pfeerick, 2023-12-21:
I'm not seeing this on another radio (T20), which should be behaving exactly the same as the Pocket. What I am seeing is the inability to actually change the state of the checkbox for the global functions, which has just had a fix done.

closed 2023-12-21 by pfeerick