EdgeTX/edgetx #4416: No LUA mixes on BW radios?

prikrylm, 2023-12-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm missing lua mixes on BW radios. I can see this feature on color radios, so using some different build options solve my whish? Is there some simple way to get LUA mixes on BW radios? (I'm using RM Pocket and MT12)

I'm using LUA mixes for ABS (for example) on OpenTX surface radio, but new radios as MT12 has no OpenTX support, so I need LUA mix.

### Describe the solution you'd like

Prebuild firmware with LUA mix page on BW radios. Maybe build by params set on Companion as it was on OpenTX?

pfeerick, 2023-12-13:
As has been done in 2.8 and 2.9, once the release version for 2.10 is out, there will be custom versions of firmwares generated at https://github.com/pfeerick/lang-firmwares for each radio for each of the languages, as well as certain other options like LUA_MIXER if it there is enough space left to turn on that feature for a given radio. We can also add it as an option to the cloud-builder on Buddy so that it can be enabled on demand there (which in the future will become available in Companion).

prikrylm, 2023-12-13:
Great prospects, thanks. :)

closed 2023-12-13 by prikrylm