EdgeTX/edgetx #4417: Em crash on SD screen (F4 BW radios) since #3846

3djc, 2023-12-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Flash BOTH firwmware and bootloader to nightly
Start radio, go to SD browser screen : crash

Confirmed on T20/Pocket/Boxer
Not existing on X7 (f2)

### Expected Behavior

No crash

### Steps To Reproduce

Flash BOTH firwmware and bootloader to nightly
Start radio, go to SD browser screen : crash

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster Boxer

pfeerick, 2023-12-13:
This may not be entirely restricted to F4... I just tried LR3PRO (STM32F205xE) and it seemed fine, the TX12MK2 and T20 did crash... *and* so did the TX12 (STM32F205xE)... I tried against a different SD card just in case it was our old SD timing issue rearing its ugly head but that made no difference. I'll try bisecting to make sure #3846 isn't getting unfairly blamed.

pfeerick, 2023-12-14:
... and now the T20 has stopped crashing (with latest nightly)... so now I'm wondering if #4396 fixed that 😕 🤪

3djc, 2023-12-14:
I haven't tried T20 this morning, but using todays nightly, Boxer, pocket and MT12 do crash (all using different sd). Going to try t20 now

3djc, 2023-12-14:
T20 still does it here with todays nightly

philmoz, 2023-12-14:
Flashed latest main branch firmware and bootloader to MT12.

Radio does not crash entering the SD menu.

However when I startup into the bootloader and plug in the USB cable there is no SD card connection to my Mac.
Reverting to the original bootloader restores the SD card function.

3djc, 2023-12-14:
Did you self compile or used GH nightly?

philmoz, 2023-12-14:
Selfbuild.

philmoz, 2023-12-14:
Flashed GH nightly (both FW and bootloader) to MT12.

No crash on SD menu.

No USB SD drive in Bootloader.

3djc, 2023-12-14:
Would you mind trying the nightly downloaded from here, trying to assess if that makes any difference ?

philmoz, 2023-12-14:
Just did that - see previous message.

pfeerick, 2023-12-14:
I think there is some other factor that is not strictly firmware build
relates going on here... As when I tried before flashing the latest
nightly, the T20 initially didn't crash, but did after a restart, and
consistently did it from then onwards. And then after DFU flashing it
refuses to crash now. However, the Zorro that was crashing the other day is
still crashing with latest nightly. My Boxer refuses to crash (which is
annoying as I could then attach to the debug header). Neither T20, boxer or
zorro have issues with SD access during bootloade - well, the boxer did do
a device not recognized one time, but a reconnect of the USB fixed that.
Only bootloader issue I'm seeing may be that exiting doesn't start the main
firmware?

3djc, 2023-12-14:
Ok, the fact it does affect bootloader mode narrows down things, but unfortunately, my boxer seems to be the only one persistently crashing, but I don't have a hardware debugger. Damm

philmoz, 2023-12-14:
I tried building various commits from the main branch to see where the bootloader USB SD stopped working.

There does not appear to be any commit in Git where the bootloader USB SD works on my MT12???

The earliest commit that works at all is ca53550c (PR 4277). Prior to this the radio will not enter bootloader mode at all - just goes straight to firmware even with trim switches pressed on power up.

All the commits I tried from PR 4277 onwards have the USB SD issues. The version that came with the radio (dated 30th Oct 2023) works. Perhaps some interim init code did not make it into Git?

Also at one point I got the SD tab crash with a build from mid November. After a few restarts it went away. So this does not seem to be related to a specific build and is very random.

swissfreek, 2023-12-15:
My MT12 also crashes when I page over to the SD files tab in the System menu. Every once in a while, it actually doesn't crash, say... 10% of the time? Less?

Actually it's funny, it was just doing it less than five minutes ago, and as I was typing this I went into the Sytem menu and went back and forth a few times and it got to the SD screen without crashing... 🤔

swissfreek, 2023-12-15:
> I tried building various commits from the main branch to see where the bootloader USB SD stopped working.
>
> There does not appear to be any commit in Git where the bootloader USB SD works on my [...]

Oh, I didn't realize the MT12 bootloader thing was a bug, I figured it was just different on this radio and I was pushing the wrong buttons. I was not able to get into the bootloader with the radio fresh out of the box, which was a late October build (before PR 4277 I assume since that's early November).

I also get the crashes at the SD screen, except maybe 10% of the time it will get to that screen without crashing. And in fact as I was typing this I was paging back and forth trying to get it to crash, and all of a sudden it stopped crashing (but five minutes ago it did, and I haven't loaded new firmware or anything). I didn't experience a crash until I started loading nightly builds last weekend. So out of the box I didn't have that issue.

pfeerick, 2023-12-16:
On the MT12, it's the top two trims (T1 & T2) inwards. I think it was
something different early during firmware development, but didn't stay that
way for long.

swissfreek, 2023-12-16:
Well I'll be damned, that did work. So my initial theory was correct, I'm just dumb 😜 Wish this was documented somewhere. I guess now that we have EdgeTX Buddy (which the manual recommends and turns out is pretty great) it's less of a "thing", but I spent a good amount of time trying to do it the old-fashioned way at first.

pfeerick, 2023-12-17:
Nah, just some radio have to be different... just wait 'til you use one that needs you to press a BOOT button before DFU works 😆 ... especially when it's a concealed BOOT0 button 🤣

3djc, 2023-12-18:
Issue understood, fix coming soon

jims1234, 2023-12-19:
3djc has been very helpful, I'm just an end user tryomg tpo use the Pocket for the 1st time .. I've posted several screen shots of the issue as its impacting me in the RCGroups.com Radiomaster POCKET thread. Thought I had it licked this morning by moving  back to the factory firmware but that was premature.. The same issue came back. On the Pocket all seems to work Ok till I try to display the Radio Page 2 of 7.. when I do the radio does not crash but the screen blanks for .5 seconds then displays the last selected MODEL. I just has this e-mail fromRadiomaster Support and this was my reply below in ""..

"Thanks for the quick reply but my issue is sadly different. I read the github string 
Mine is an ELRS radio by the way with an RM 4in1 in the nano slot.. I turned BOTh OFF and removed the NANO during this.

I’m already running  that release that you sent me as an attachment  and it’s an intermittent issue since initially That firmware FIXED it but now the lockup is still re-occuring with that RM Factory Version 1fdb58ba  2023-07-27 15-24-19 installed..
I think I need another POCKET radio, I notified banggood.com but they did not reply   More as it evolves
Thanks
jimS

From: RadioMaster Support Team <[support@radiomasterrc.freshdesk.com](mailto:support@radiomasterrc.freshdesk.com)> 
Sent: Tuesday, December 19, 2023 1:56 AM
To: [jims123@gmail.com](mailto:jims123@gmail.com)
Subject: Re: Can't view the files on the SD CARD using Page 2-of-7 yet other Pages load and work Ok

Hi Jims123,

This is EdgeTX software problem, not just Pocket, black and white and have all sorts of the firmware of the remote control of the screen the same phenomenon, now I see here is trying to a solution, I think in a few days should be able to fix the problem: https://github.com/EdgeTX/edgetx/issues/4417
You can use this firmware first and can access the SD card normally

Kind regards  "

closed 2023-12-20 by pfeerick

pfeerick, 2023-12-20:
> when I do the radio does not crash but the screen blanks

FYI, if this happens, and you have an exclamation mark `!` shown on the screen (left side) then it actually DID crash. B&W radios don't behave the same as colorlcd radios - they don't halt the UI completely as it is a lot simpler. You will be able to flash this firmware with tomorrows nightly, and this should resolve the crash if you enter the SD manager page... (i.e. page 2). You can of course navigate the menu in the other direction (page 7, 6, etc) and can get to page 3 without it crashing.

philmoz, 2023-12-20:
Is the missing USB SD drive in the bootloader fixed as well?

pfeerick, 2023-12-20:
No. I'll open a new issue to track that as this one was tracking the EM on the SD card screen which is now resolved. The other issue appears to be somewhat MT12 specific - I can repo it with the MT12, but the TX12, Zorro and Boxer don't have that issue ... which is interesting as the MT12 is closely related to the zorro and boxer.

PR #4436 (2023-12-20): fix(bw): Uncleared reusable buffer in sdmanager
Some bugs can stay hidden for years .....

Did a bit of housekeeping while I was at it

This fixes #4417