EdgeTX/edgetx #4421: Incorrect name of 6-position switch in Russian translation

Shouldercannon, 2023-12-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

In the companion with the Russian language, the 6-position switch has incorrect position names.
[image]

### Expected Behavior

In the companion with English, the 6-position switch has the correct position names.
[image]

### Steps To Reproduce

[image]

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Win10 x64 22H2

kobakirill, 2023-12-22:
Fix 6p.
 https://github.com/EdgeTX/edgetx/pull/4451

closed 2023-12-23 by pfeerick

Shouldercannon, 2024-01-22:
The translation problem has not been resolved. Now it is in version 2.9.3.

pfeerick, 2024-01-22:
The fix is in 2.10, not 2.9.3. I can add it to the 2.9 branch though, and it will be present in 2.9.4 if there is a future release.