EdgeTX/edgetx #4422: Issues after remapping Hardware Switches

mrac1, 2023-12-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After remapping the Switches SH <-> SF (SH is now 2-was and SF is now Toggle), the Preflight Checks Switches are displayed incorrectly. SF = 0 and SH is without flags until you exit and re-enter the menu. Then SH is correct but SF remains at 0. From time to time, switching to the Preflight Checks menu also causes crashes with Emergency Reboot.

### Expected Behavior

SF = SF with Flags + SH = SH with Flags

### Steps To Reproduce

Swap Switch SH and SF in the Hardware Menu, goto Preflight Checks.

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

-

wombats-kode, 2023-12-30:
I have a related issue where I have also swapped SH to a two position toggle that I use to arm/disarm all models.  Preflight check does not permit high/low flags selection on SH switch, only option is high which in my case is 'armed' so I have to disable completely.  Would be great to have ability to select SH(low) to prevent inadvertant arming when changing model.  Thank you

3djc, 2023-12-31:
Not sure I understand the issue. if you swap switches, there is no way for the radio setup to adapt itself. You need to first change your switch settings in hardware screen to match the new physical implementation, you need then to check your model  settings to take into account those changes, this is to be expected. In your case, preflight checks need to be redefined to match hardware. Tested on 2.10, cannot find an issue with it

wombats-kode, 2023-12-31:
I have remapped the hardware to account for the change of switch role, I guess that is why I am having an issue as it knows it is now a two position switch but preflight checks only allows me to select one position to display a warning. The other option is no warning.  Just so happens I would prefer to set the warning for the other switch position.  This is on v2.9.2.

3djc, 2024-01-01:
I'm sorry but this is inaccurate:

Photo taken with 2.9.2

[image]

[image]

[image]

mrac1, 2024-01-01:
Set SF to Toggle an the issue appears. You can also trace the error in the Companion.
[image]
[image]

3djc, 2024-01-01:
I was responding to [wombats-kode](https://github.com/wombats-kode)

As far as your issue is concerned, I do not understyand what your issue is

philmoz, 2024-01-01:
> Set SF to Toggle an the issue appears. You can also trace the error in the Companion.

@mrac1 this has been fixed in 2.10.

mrac1, 2024-01-01:
Perfect, thank you!

closed 2024-01-01 by mrac1

wombats-kode, 2024-01-01:
> I was responding to [wombats-kode](https://github.com/wombats-kode)
> 
> As far as your issue is concerned, I do not understyand what your issue is

@3djc It looks like your version is doing what I need.  I do not get the option of setting two states for SH, I just get the option to set SH(arrow down).  I get no option to set SH(arrow up).  Hardware is set as you indicate in the image and flag debug show the states switching correctly.

I do also get the problem where entering pre-flight checks causes the radio to crash regularly.