EdgeTX/edgetx #4424: Companion update function in 2.9.2 does not work in Linux Cinnamon

dlabrie55, 2023-12-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When Trying to download updates I get the  error... Unable to download Meta Data. Error:6
SSL handshake failed

### Expected Behavior

Multiple updates should download, decompress and install. 

### Steps To Reproduce

1. Open Edge Companion 
2. Click on the Download Components arrow.
3. Error messages occur for all downloads.

### Version

2.9.2

### Transmitter

FrSky X9D+

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.2 Cinnamon

### Anything else?

[image]

dlabrie55, 2023-12-18:
After getting advice to download [http://archive.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1f-1ubuntu2.20_amd64.deb](https://github.com/EdgeTX/edgetx/issues/url) I was able to download and install all fo the data.

pfeerick, 2024-01-12:
Duplicate of https://github.com/EdgeTX/edgetx/issues/4316

PR #4533 (2024-01-14): 2.9 : Linux CPN AppImage SSL packaging
Fixes #4424, fixes #4316

Summary of changes:
- EdgeTX/build-edgetx#16 added two packages, as well as a hint for the build system as to where SSL libraries located, and I missed that when bringing #4270 over to 2.9 via https://github.com/EdgeTX/edgetx/pull/4309 (which was needed due to upsteam changes). Should only need one of those packages and the hint for 2.9... it appeared to detect locally... need GH to repo that now. Updating edgetx-dev would be the better fix, but this is out of sync with the state of main. 

~~edit: Looking at the output file, loo [... cut]
closed 2024-01-14 by pfeerick