EdgeTX/edgetx #4427: GPS value poor displayed on Jumper T20

theysp, 2023-12-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

GPS value in telemetry is not well displayed:
[image]

### Expected Behavior

Bellow is the same GPS value of the same quad now displayed on tx12.
[image]

[image]

### Steps To Reproduce

1. connect to quad
2. go to telemetry screen in its model
3. discover new
4. look for GPS row

### Version

Nightly (Please give date/commit below)

### Transmitter

Other (Please specify below)

3djc, 2023-12-17:
Have you tried a 2.10 nightly on tx12 ? Because they run the same code

theysp, 2023-12-18:
> Have you tried a 2.10 nightly on tx12 ? Because they run the same code

Thank you, I would try it on tx12, and go back to you this weekends.

theysp, 2023-12-22:
> Have you tried a 2.10 nightly on tx12 ? Because they run the same code

Unfortunatly, this nightly build cannot work on my tx12. After flashing, it went dead, cannot respond to interactivity such as pushing buttons or so.

[image]

3djc, 2023-12-22:
Seems to be working fine here

[image]

theysp, 2023-12-23:
I am now using SD cards contents from edgetx 2.9.0 from EdgeTX Buddy.

The previous not working tx12 (not mk2, just tx12) was with 2.4.2 SD card contents. After I tried 2.9.0 SD card contents, it failed to boot with NO SD CARD warning on screen. Where can I get the correct SD card contents for 2.10.0-nightly?

Is the formating of GPS info related to the contents in SD card?

By the way, I got the detailed info of the firmware on t20 as follows:
[image]

3djc, 2023-12-23:
sd card content is irrelevant, it doesn't affect GPS display in any way. You need a more recent nightly to get rid of the sd card error

theysp, 2023-12-24:
> sd card content is irrelevant, it doesn't affect GPS display in any way. You need a more recent nightly to get rid of the sd card error

It's ok to leave tx12 where it was. I flashed edgetx 2.9.2 to tx12, the GPS info is displayed well. If nothing has changed since then, the error may be caused by the radio jumper t20 itself.

closed 2023-12-24 by 3djc