EdgeTX/edgetx #4439: MT12 - No USB SD at bootloader

pfeerick, 2023-12-20:
Flashed latest main branch firmware and bootloader to MT12.

When I startup into the bootloader and plug in the USB cable there is no SD card connection to my Mac.

Reverting to the original bootloader restores the SD card function.

_Originally posted by @philmoz in https://github.com/EdgeTX/edgetx/issues/4417#issuecomment-1855433539_

pfeerick, 2023-12-20:
I can also reproduce this with my MT12 on Windows... A USB drive is detected but not mounted. If I use the USB storage option from the main firmware, that works. I tested with a github main build from two days ago - https://github.com/EdgeTX/edgetx/commit/87e71d0a33bafd3ea7b009f1c44a602665871d17. 

Note... https://github.com/EdgeTX/edgetx/commit/ca5355e0c7e12885d1f8a7eeb9dfee6c94df9982 is the PR that added MT12 support, so before that it didn't even exist in the repo ;) So it's no surprise it wouldn't enter the bootloader prior to that PR - plus the trims keys are moved around IIRC, so while you could flash maybe a zorro firmware on it, the bootloader keys would be wrong.

3djc, 2023-12-20:
ok, I'm a bit puzzled here. if there was a widespread 'no sd in BL' issue, we should have heard strongly about it, yet we havent.

I just tried the factory supplied image :https://github.com/EdgeTX/edgetx.github.io/blob/master/downloads/MT12-PROD-28-10.bin?raw=true on two MT12, both give me SD content on windows and macos sonoma. (tested using a DFU flash from companion). The only odd thing there was that radio name was not appearing, will look into that.

As for current main, yes, I agree there is an issue

raphaelcoeffic, 2023-12-20:
Besides.... It works here with latest code from `main`.

3djc, 2023-12-20:
@pfeerick @philmoz With what SD did you try ?

philmoz, 2023-12-20:
> Besides.... It works here with latest code from `main`.

Still not working for me.

philmoz, 2023-12-20:
> With what SD did you try ?

The one that came with the radio.

I can't see how it could be the SD card:
- USB SD in firmware works (both original and current main)
- Bootloader from original firmware works
- Bootloader built from any GitHub commit fails.

3djc, 2023-12-20:
Could you try with a big brand one ?
(bootloader doesn't use DMA, thats why it is different than firmware)

philmoz, 2023-12-20:
I don't have any spare SD cards at the moment.

Why would the original firmware work; but GitHub version fail?

raphaelcoeffic, 2023-12-20:
✅ Tested right now on Linux:
[image]

raphaelcoeffic, 2023-12-20:
> Bootloader from original firmware works

I believe that one had still the old SD/SPI code. Timings might be slightly different (not much though). The SD cards might be very much boarder line....

3djc, 2023-12-20:
> I don't have any spare SD cards at the moment.
> 
> Why would the original firmware work; but GitHub version fail?

Because the original firmware is much older than the GH merge, usually 1 or 2 month, but more in this case since MT12 have been long in the making

3djc, 2023-12-20:
Just put your TX16S in there, all you want is see data

pfeerick, 2023-12-20:
Stock MicroSD card in my case. I tried two different MicroSD cards, and it did mount with those. So given this is *only* the MT12 that I've seen do this to date, and JCs comment just now, I'm inclined to say it's a MicroSD incompatibility issue. Yet another reason to get XTSD onboard storage happening...

philmoz, 2023-12-20:
I expect that will cause a few support issues.

pfeerick, 2023-12-20:
SD storage always has... that's why there are a lot of people who just replace the SD card that comes with their radio even if there is nothing wrong with it... just because!

3djc, 2023-12-20:
We clearly need to address that, because those SD card are "acceptable", so they should work

raphaelcoeffic, 2023-12-20:
> We clearly need to address that, because those SD card are "acceptable", so they should work

Someone up for a logic trace of what is happening? My SD card is working, so that won't be me this time. Besides, it seems to be working in the firmware, so it seems the card is pretty touchy on timings, which it should really not. The only difference here is the use of DMA in the firmware, whereby we don't in the bootloader (size constraint).

3djc, 2023-12-21:
Ok, bit of an update. This is a pure nightmare. Some cards that did work before don't anymore since #3957. Radio model doesn't matter, cards do.

It is SO sensitive to timings that those non working actually work when compiled with debug and attached to SEGGER.

We will try to take SPI traces to see if we can spot something

raphaelcoeffic, 2023-12-21:
> Some cards that did work before don't anymore

It is to be noted that only cheap *no name* cards are affected. For my part, I don't have a single affected card (but a lot of cheap *no name* cards).

Also, this is only about SPI, which is not really the main use-case for SD cards (only a legacy compatibility mode that is not very well supported by all vendors).

gagarinlg, 2023-12-21:
Is the clock polarity correct?

PR #4450 (2023-12-22): fix: increase init repeat max time for slow(est) cards
Sometime, the fix looks trivial, yet the journey to get there hasn't been ....

This fixes #4439 

@raphaelcoeffic co-authored this change
closed 2023-12-22 by pfeerick