EdgeTX/edgetx #4440: Mt12 pop up when added trim

TB2185, 2023-12-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Very confusing when adding Trim to what you’re actually adjusting via steering or throttle trim or even weight. It took me some time to configure the radio to have a Pop up for what I was Adjusting.  This should be standard to make it easier for the ended user.  Planes we know what trim does what but with a car radio they all normally have a pop up screen that says ST or TH

weight should be re named to ST DR and TH DR  DR standing for Dual Rate..  happy to help go through anything thing else. Been into Rc racing for 20+ years. Happy to have an inexpensive open source platform. 

### Describe the solution you'd like

See above.

fiechr, 2023-12-23:
The popup makes sense, but I'm not sure if it is a good idea to rename it.

Personally, although I'm running surface and air models for years, I think the term "weight" makes much more sense than "dual rate". Even if someone has no clue what anything of this means, it's more likely to be understood because of the name alone. And let's face it, beginners have no clue what "dual rate" on their radio means, anyway.

For intermediates or pros, I think, it's not really a hurdle to understand that weight is basically the same as dual rate and work with that. Weight is a very fundamental term in EdgeTX and I think it should be used for surface, too. But this is just my personal opinion.

Ideal, if possible, would be some kind of hint or help text that notes that weight and dual rate is basically the same thing.

3djc, 2023-12-23:
If you want this displayed you can enable it in model settings. It so pop up, you can make it a temporary or permanent display

pfeerick, 2023-12-23:
Dual rates isn't really valid either when you have more than two rates 😆

TB2185, 2023-12-23:
> Dual rates isn't really valid either when you have more than two rates 😆

99.9% of people don’t use more than two rates on a surface radio. It’s not like an airplane.

TB2185, 2023-12-23:
> If you want this displayed you can enable it in model settings. It is not a pop up, but you can make it a temporary or permanent display

Yea took me a few hours to figure out how to do it.  This should come setup and turned on.  Why be the only radio on the market that doesn’t have these labels. Same with the weight vs Dual rate. Even planes call it dual rates..

Wandereradi, 2024-04-09:
@TB2185 I would like to have the steps to create pop up when i do a trim change to any channel. I second your thought on the way this radio is set up,