EdgeTX/edgetx #4441: [X10Express] strong screen flickering

ParkerEde, 2023-12-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

with the current nightly version for the X10Express I now get a strong screen flickering.

### Expected Behavior

there should no screen flickering

### Steps To Reproduce

1. flash current nightly on X10 Express
2. PowerOn and observ Screen

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### Anything else?

current nightly from today

pfeerick, 2023-12-20:
Please try https://github.com/EdgeTX/edgetx/pull/4433 or tomorrows nightly.

ParkerEde, 2023-12-24:
yes, it is okay now. Thx

closed 2023-12-24 by ParkerEde