EdgeTX/edgetx #4444: MT12 Pre-Start Check Switch-Warning: Switchpos not selectable

Tiefflieger68, 2023-12-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Radio MT12 
Issue: Pre-Start Check Switch-Warning: it's not possible to select or disable Switchpos by short press enter key. Only by long press enter and auto detect switch pos.

### Expected Behavior

select switch A (there is only switch A at MT12), short press enter to rotate pos up, midd, down and off. Works fine at X9 lite (etx 2.9.1)

### Steps To Reproduce

open model menue SETUP (2)
find "Pre-start Checks" and enhance submenue
select S-Warning - nothing happens by short press enter

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### OS Version

2.10.0 nightly de0379c2  2023-12-20

pfeerick, 2023-12-20:
There are four switches on MT12 - A (3pos under wheel), B (behind the wheel at the top of the cavity, C (on the grip) and D (at the base at the back). If they are not all showing for you, something is wrong with your SYS -> HARDWARE -> switches configuration. 

With a nightly from two days ago, I can use short press after selecting the switch warnings line to select any of the four switches and then short press to alter between required state (or ignore). Long press on the switch warnings line iscaldo doing the auto detection. 

I will check again later with current nightly just in case something changed though.

philmoz, 2023-12-20:
SB, SC and SD are toggle switches so they should not be selectable in the pre-start checks.

pfeerick, 2023-12-20:
Yeah, should have remembered that... did fix some of that up! Looks like the defaults need updating... Mt12 reset to 3pos for B-D with a new card. 

OK, I can repo this when there is only one valid switch in the list.

pfeerick, 2023-12-20:
The `-1` appears to be the offender here, but that then also sets off the full line selection if it's just removed. 

https://github.com/EdgeTX/edgetx/blob/6b1256db6651b5e80c93d28e0f1c84572d2f3cdd/radio/src/gui/128x64/model_setup.cpp#L317

PR #4448 (2023-12-21): fix(MT12): Cannot properly edit pre-start check switch warning state on MT12.
Fixes #4444

Add some special case logic to handle when there is only one switch available to trigger a pre-start warning.
closed 2023-12-21 by pfeerick