EdgeTX/edgetx #4452: Allow analog sources to be used in things like expo, weight

3djc, 2023-12-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

To be able to drive things like expo or weight using a pot, you need to setup a SF that copies the value in a gvar, then use that gvar as source. This seems cumbersome

### Describe the solution you'd like

Ability to use analog sources in weight and expo (possibility others, havent investigates yet)

### Describe alternatives you've considered

use gvar

philmoz, 2023-12-24:
I think this could be done, where the weight/offset/expo etc could be either a value or a 'source' (which includes gvar's so it becomes a superset of current functionality).

For Mixes and Outputs the existing data structures look big enough to work; but Inputs (ExpoData struct) would probably require an extra byte added - 64 bytes more RAM needed for model data.

3djc, 2023-12-24:
yes,  clearly the way to go. A bit messy since the gvar/value will need to be reworked; but yes

raphaelcoeffic, 2024-01-01:
> 64 bytes more RAM needed for model data

Probably affordable 🤞

3djc, 2024-01-01:
Given the benefits, I would agree

philmoz, 2024-01-06:
I started to look at this; but found one complication.

Currently weight/expo/offset etc supports negative (inverted) GV's as source.

However a normal 'Source' does not currently support inverted/negative values - switches do; but not sources.

To implement this without losing existing functionality we would need to allow the source to be inverted.

3djc, 2024-01-06:
That would be a nice addition actually

philmoz, 2024-01-06:
> That would be a nice addition actually

Unfortunately it's going to require a lot of work - the code assumes source values are unsigned so much of the code needs to be tweaked.

I think it would be best to first do the inverted source feature as a separate P,R so it can be reviewed and tested, then revisit this one.

3djc, 2024-01-06:
Makes a lot of sense

PR #5220 (2024-07-26): feat: use any source for expo/weight/offset etc
Fixes #4452 

Time to continue with this change, now that source invert has been implemented.

Allows any source to be used in inputs and mixes for fields like weight, offset, expo, etc (not just GVars).

TODO

Radio:

- [x] Data structures
- [x] Save and load to YAML files
- [x] Mix calculations
- [x] Color UI
- [x] B&W UI
- [x] Fix tests

Companion:

- [x] Data structures
- [x] Save and load YAML files
- [x] UI for settings fields
- [x] Binary file conversion
- [x] Testing
closed 2024-07-26 by pfeerick