EdgeTX/edgetx #4453: Ability to use 'unactive' trims as 3 way switches

3djc, 2023-12-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

when a trim is disabled in FM, it cannot be used as a 'switch source'

### Describe the solution you'd like

Ability to use it. There are questions to be thought through as in essence, that switch wont be always existing, what do we do when it doesnt

### Describe alternatives you've considered

put more mixer lines instead of a simple one

philmoz, 2023-12-22:
Where are you seeing this?
I've been using T5 & T6 as switch sources in my TX16S since 2.8.

3djc, 2023-12-22:
B&W. Can you do a mix line with a trim switch (NOT the trim channel) a a source ?

3djc, 2023-12-22:
Just checked, you can use trim channel, like on b&w. I'll give you an example so it is clearer. You want to disable the trim in flight mode, and then use the trim switch as a 3 pos switch to drive a winch a 4x4 car for example. As source it would give 0 untouched, and either -1024 or +1024 depending on the direction

pfeerick, 2023-12-23:
This is basically a duplicate of https://github.com/EdgeTX/edgetx/issues/1741

philmoz, 2023-12-23:
[image]

philmoz, 2023-12-23:
Potential fix - radio/src/opentx.cpp, getTrimValue function.

Change:
```c
    if (v.mode == TRIM_MODE_NONE) {
      return result;
    }
```

To:
```c
    if (v.mode == TRIM_MODE_NONE) {
      uint8_t tidx = idx * 2;
      if (trimDown(tidx)) return -1024;
      else if (trimDown(tidx + 1)) return 1024;
      return 0;
    }
```

pfeerick, 2023-12-23:
Yes, you can use a three-line mix, with MAX, 0, and MIN as the sources (i.e. as JC wrote under "Describe alternatives you've considered"), but that isn't anywhere as end-user-friendly as just being able to assign a switch as a source in a single mix line. You just can't use it _directly_ as a source though, which is the focus here... 

And regarding the switch not existing... from a usability standpoint, shouldn't that result in the channel being 0% / 1500us... until the user fixes it up? I guess UI wise, for colorLCD at least maybe it could be shown in the WARNING colour, to indicate an invalid source? Maybe B&W could show square brackets around invalid source lines, or even just blink invalid lines... something simple.

PR #4465 (2024-01-05): feat: Use trim as a 3 pos toggle switch source
Fixes #4453 

If trim is disabled in flight mode, treat it as a 3 position toggle switch in 'getValue' function.
closed 2024-01-05 by pfeerick