EdgeTX/edgetx #4456: [MT12] Default channel order, naming and default trims for these

fiechr, 2023-12-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I understand that the default channel order for this surface radio was initially TH-ST, although it can be changed to ST-TH in the sys menu, the default setting seems to be ST-TH now, which is good!

I still would recommend some considerations and/or optimizations for this, though.

### Describe the solution you'd like

* At the moment the user can switch between order options named "ST" and "TS", which is a bit inconvenient, since steering is also known as "ST". I would recommend using something as "ST-TH" and "TH-ST" for this option (two letters instead of one, since that's what surface users are accustomed to, while in the air "domain", user are used to only one letter naming (A,E,T,R), in the surface "domain" they use two letters (ST, TH) for ages.

* The next question would be, if this option is _necessary at all_, since I've never came across a radio or recevier that uses another than channel order than ST-TH (CH1: Steering CH2: Throttle). There are some very rare exceptions (for examples some WLToys models), but the user can always change the channel order anyway. As I understand, this is just a convenience thing for newly created models.
Personally, I would simply get rid of it and set the default channel order for newly generated surface models always to "ST-TH". This is an air "thing", because there are these different modes and channel orders for air, but for surface it's much simpler -- and why make it more complicated again?

* The trims "T1" is named "TH" and the "T2" is named ST, which, I assume goes back to the initially different channel order. I would recommend to rename this to T1: "ST" and T2: "TH" (it's not labeled on the device itself, device uses "T1" to "T5". Or name it to "T1" and "T2". But the current names are misleading, especially when the correct channel order is selected.
Personally, I would name them "T1" and "T2" but still use T1 for steering trim and T2 for throttle trim per default. This is consistent with the device labeling and makes sense on a logical level, too.

pfeerick, 2023-12-23:
I'll just comment on that last point... trims inherit the name of the axis they are paired with... so if you call the TH axis  A, the trim for it will be named A. 

I generally agree with the second point, in that we probably don't need that setting at all... especially when for the odd model you probably need to set up a completely custom mix order anyway. Although that does go against the principle of giving the user _options_...

3djc, 2023-12-23:
Isn't default channel order fixed already? And I fail to see the purpose of removing the setting, if you don't want to use it, then simply don't

3djc, 2023-12-23:
I'm wondering if we should have a settings for trims display, that would default to only the first two for surface radio

fiechr, 2023-12-23:
> Isn't default channel order fixed already? And I fail to see the purpose of removing the setting, if you don't want to use it, then simply don't. If you for example only use those wltoys, then you will want to change default 

The default for many WLToys is throttle on CH2 and steering on CH4. Which means it needs to be changed later anyway.

In my opinion this default setting makes sense, if there are multiple commonly known values that people often use.

fiechr, 2023-12-23:
> I'll just comment on that last point... trims inherit the name of the axis they are paired with... so if you call the TH axis  A, the trim for it will be named A. 

I've renamed the source in the hardware page and the input in the input page, but the names of the trims didn't change, at least not in the input settings where I select them. Or did you mean that it's convention to name them this way? Or something completely different?

Maybe it should also be considered, that the trim buttons on this surface radio are not placed nor labeled in a way that makes them obviously belong to a certain axis or control (like below or besides a certain stick). The device itself only labels them as T1 to T5. I therefore would tend to name them the same in the software to avoid confusion. At least if the whole naming thing is fixed as it appears to me at the moment.

jakezilla, 2024-01-24:
Has there been a decision on renaming the trims in the firmware to match the physical labels on the switches? It seems to be a source of confusion fairly regularly in the support groups.

3djc, 2024-01-25:
The trim name matches the axis they trim for the first 2 (so TH or ST), then trim number, as any other EdgeTX radio

fiechr, 2024-01-25:
Yeah, but the correct order is ST and TH. The default order settings doesn't change that. If you look at the radio from the right side, you see that they are labeled "T1", "T2" and so on. One would expect that the first one defaults also for the first channel. I always mix up the trim switches with this radio because of that.

BTW. the same is true for the Drive Mode setting, where the trims can be disabled or set to different mode values, the first one should be T1, the second T2 and so on, but they aren't.

fiechr, 2024-01-28:
Maybe I've described it wrong or got it wrong. Here is what I mean:

If I press this trim switch,
[image]
then I get a reaction on the "T2" labeled control in the debug screen:
[image]
Vice versa, if I press the T2 button, the T1 labeled control shows some input value.
All the other trims (T3-T5) react as expected.

This is how my hardware page looks:
[image]

In the Flight..., pardon me, Drive Mode page, it's the same:
[image]
What is shown here in this screenshot above, enables only "T2", while "T1" is off. All others are in the correct order. For example the last one does in fact enable/disable "T5".

The "Trim" setting set to "Yes" (or any other value, for that matter) in the Inputs page does not change what is shown above!

Of course, they could be labeled and mapped completely different, but in my opinion, if there's no important reason, it makes the most sense to stick to the labels on the hardware. This would mean labeling them T1 and T2 and map/set them per default to Steering and Throttle (in that order).

BTW, on the Flysky NB4, "TR1" also controls steering trim, while "TR2" controls throttle trim -- at least in the default settings, of course, it can be changed.

These are the other trim buttons on the MT12 (just for completeness):

[image]
[image]
[image]
[image]
[image]

3djc, 2024-01-28:
Those are close to invisible on my MT12, but fixed those in #4582

Could you test on new blank SDcard ?

ST should be on CH1, modded by T1
TH should be on C2, modded by T2

My understanding is it in this PR how it should be

fiechr, 2024-01-28:
> My understanding is it in this PR how it should be

I agree, it now works as expected. I used a freshly formatted SD card only with the latest nightly (EdgeTX) stock contents and sound files.

Debug keys output is now working as expected and the trim order in the Drive Mode page is now in the correct order, too.

Thank you for your work BTW! :)

tonyvsuk, 2024-03-18:
I've been setting up my new radio and stumbled across this ticket.

Is this in the 2.10 release candidate?

3djc, 2024-03-18:
yes

closed 2024-03-18 by 3djc