EdgeTX/edgetx #4459: MT12 - no LEDs at handle

Tiefflieger68, 2023-12-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The LED's in the handle stays solid blue. Settings in SF have no effect.
Going back to stock firmware everything is fine

### Expected Behavior

LED color can be set by SF

### Steps To Reproduce

load the first model of stock and move the throttle stick. 
with stock firmware everything is fine, with 2.10.Nightly 02e2be91 LED stays blue

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### OS Version

02e2be91

pfeerick, 2023-12-24:
Can't reproduce with a configured model (I don't have any of the stock ones on atm). Are you restarting the transmitter after making changes to the RGB led special functions? Also check if you have the latest SD card pack, as the RGB led scripts have had some changes - shouldn't really affect the solid colours (other than perhaps brightness brightness) but would mess with any animations.

3djc, 2023-12-24:
Not seing it either, works fine here on latest builds

3djc, 2023-12-24:
Try this script setup as shown:
[image]

[thr.zip](https://github.com/EdgeTX/edgetx/files/13761378/thr.zip)

Reboot radio, your led should match the status of throttle position

Tiefflieger68, 2023-12-24:
hm - strange...

created a new model and added the SF entry
your script works well if stock firmware (MT12-edgetx.2.10.0-2023.10.30.bin) is flashed
[image]
If  one of these firmware is flasched the LED stays solid blue.
Flashing back to stock firmware, the LED works well again

I've downloaded the Nighlys directly from here:
https://github.com/EdgeTX/edgetx/releases

3djc, 2023-12-24:
[image]

[image]

Are you doing a full flash with dfu ?

Tiefflieger68, 2023-12-24:
my version screen shows exactly this.
I've flashed by coping the FW-File to SD and flash in bootloader mode

give me some minutes to do some tests...

Tiefflieger68, 2023-12-24:
I was wondering that in "Radio Setup" menue "Custom Scripts" was unchecked. So I've activated this. After this my SF entry was deactivated. After activating the thr.lua worked well. but not the other Luas like "red".

After powercycling the radio the entry "Custom Scripts" is unchecked again.
But my radio.yml seems looking well

[radio.yml.txt](https://github.com/EdgeTX/edgetx/files/13761994/radio.yml.txt)

fiechr, 2023-12-25:
I've flashed the same version shown above (fbe06b36) and the LEDs in the handle work as expected.

This is the file I compiled: [edgetx_mt12_fbe06b36.zip](https://github.com/EdgeTX/edgetx/files/13765932/edgetx_mt12_fbe06b36.zip)

I wrote this script to cycle through some basic colors:

```LUA
-- Color Cycle
-- LEDs will cycle through the given colors in run(), by using the given delay.

local function init()
  RGB_CYCLE_DELAY = 150           -- 150 * 10 ms = 1.5 s

  rgb_cycle_last_time = getTime() -- getTime() * 10 ms: milliseconds since radio boot
  rgb_cycle_i = 1                 -- Table index of current color
end  -- of init()

-- Set all LEDs to the given color.
local function setRGBLedColors(t)
  for i = 0, (LED_STRIP_LENGTH - 1), 1 do
    setRGBLedColor(i, t[1], t[2], t[3])
  end
end

local function run()
  local colors = {   -- Colors to cycle through.
    { 100, 0,   0 }, -- red
    { 100, 50,  0 }, -- orange
    { 100, 100, 0 }, -- yellow
    { 50,  100, 0 }, -- warm green
    { 0,   100, 0 }, -- green
    { 0,   100, 50 }, -- cold green
    { 0,   100, 100 }, -- cyan
    { 0,   50,  100 }, -- light blue
    { 0,   0,   100 }, -- blue
    { 50,  0,   100 }, -- purple
    { 100, 0,   100 }, -- violet
    { 100, 50,  0 }, -- pink
    { 100, 100, 100 }, -- white
    { 0,   0,   0 }  -- black (off)
  }

  if (rgb_cycle_last_time + RGB_CYCLE_DELAY) < getTime() then
    if colors[rgb_cycle_i] == nil then rgb_cycle_i = 1 end -- Reset table index after reaching the end.
    setRGBLedColors(colors[rgb_cycle_i])
    applyRGBLedColors()
    rgb_cycle_i = rgb_cycle_i + 1 -- Prepare for next call.
    rgb_cycle_last_time = getTime()
  end
end -- of run()

return { run = run, init = init }
```

Tiefflieger68, 2023-12-29:
Hi
back from Christmas Seasons.
How I should use the LUAs?
Put them into the folder SCRIPTS\RGBLED and start by SF "RGB leds"?
Put them into the folder SCRIPTS\FUNKTIONS and start by SF "LUA"?
Start direcly from SYS Menue, navigate to script and select EXECUTE?
or?

pfeerick, 2024-02-07:
@Tiefflieger68 Did you progress any further with this?

Tiefflieger68, 2024-02-08:
will check this later today and comming back to you

Tiefflieger68, 2024-02-08:
seems working with the lastest Nighty (mt12-a0ccea1.bin)
nothing elses changed at my MT12

pfeerick, 2024-02-08:
Thanks for checking. I'll close this issue then. You may be interested in the next nightly as it should remove the need to restart the transmitter after making changes to the RGB LED special function lines ;)

closed 2024-02-08 by pfeerick

Tiefflieger68, 2024-02-09:
it should remove the need to restart the transmitter after making changes to the RGB LED special function lines

sounds good!
Many Thanks to all the Devs for building this great software