EdgeTX/edgetx #4463: Instant Trim Does Not Trim Elevator

glieb716, 2023-12-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have Instant Trim set up on a logical switch, ANDing a momentary switch and the middle position of a 3POS. The throttle trim is disabled in the flight mode, only using FM0, REA trims set to :0. The rudder and aileron trims work fine but I cannot get the elevator trim to react to the instant trimming. Regular trimming works fine with the physical switch. 

### Expected Behavior

I expect that when the elevator stick is deflected either direction passed its middle position and the instant trim function is enabled that the elevator, like the other trims, gets set to the stick position.

### Steps To Reproduce

1. LO6: AND SC-, SE down
2. special function instant trim on source LO6
3. set SC neutral and SE (momentary) down with stick deflected
4. instant trim beep and haptic
5. all sticks besides throttle (disabled) and elevator set the trim values

### Version

Other (Please specify below)

### Transmitter

RadioMaster Pocket

### Anything else?

build is 2.10.0-selfbuild for radiomaster pocket

3djc, 2023-12-26:
As far as I remember, instant trimming is disabled for throttle trim on purpose

glieb716, 2023-12-28:
that is true, however my problem is that i am trying to trim my elevator and that isn't working. That was a few days ago on the factory RM Pocket firmware (2.10.0-selfbuild). Now on the latest nightly as of today and the issue persists. Here's settings before the nightly upgrade.
[glieb716-RMpocket-2.10.0.zip](https://github.com/EdgeTX/edgetx/files/13785206/glieb716-RMpocket-2.10.0.zip)

andrrew19, 2023-12-30:
I have exactly the same problem on my pocket ele instant trim moves throttle trim i have been trying sort this out for days its aetr mode 2

andrrew19, 2023-12-30:
I have messaged radiomaster awaiting a reply

glieb716, 2023-12-30:
> I have messaged radiomaster awaiting a reply

I did as well. Mine is also the mode 2 elrs model aetr, maybe I'll switch the input order to something else besides aetr and see if that helps. RM told me to try the nightly. as of the very newest 16 hours ago it's still not fixed. I would not suggest getting on the nightly yet if you haven't already

glieb716, 2023-12-31:
philmoz ftw. you fixed my channel monitor issue earlier today too

andrrew19, 2023-12-31:
Has anyone found away round this issue i cant get it to work correct on mode 2 for instant trim on ch 2 instant trim

glieb716, 2023-12-31:
nothing so far, ive just been trimming like normal. or theoretically you
could circumnavigate waiting for this pull request to get merged ^ and
build and flash the new firmware yourself

closed 2023-12-31 by pfeerick

andrrew19, 2024-01-01:
How is it completedthere is clearly a fault

pfeerick, 2024-01-01:
Because it has been fixed via #4491, as indicated in the completion message ;)

andrrew19, 2024-01-01:
I cant see how it is fixed it says it only works in mode 1

pfeerick, 2024-01-01:
The fix was that it was only correctly detecting which stick was the throttle stick (for instant trim purposes) when set to mode 1. Have you flashed the nightly yet? If you're not a developer or able to read code, perhaps it would be a good idea to actually try it out ;)

andrrew19, 2024-01-06:
So there is no fix for it in mode 2 basicly

pfeerick, 2024-01-06:
No, no, no, no! Please read what I wrote! 

> it was only correctly detecting which stick was the throttle stick (for instant trim purposes) when set to mode 1

Because with mode 1 (vs mode 2), the throttle and elevator sticks are swapped. Instant trim wasn't taking the stick mode into account when ignoring the throttle, so was always ignoring the T2 trim, regardless of it if was throttle or elevator. This has been fixed. 🤦 

Flash a nightly build, and test it for yourself.

andrrew19, 2024-01-07:
> ### Is there an existing issue for this problem?
> * [x]  I have searched the existing issues
>
> ### What part of EdgeTX is the focus of this bug?
> Transmitter firmware
>
> ### Current Behavior
> I have [...]
Hi have u managed to fix it i cant if so how did u do it

pfeerick, 2024-01-07:
1. Go to https://buddy.edgetx.org/#/flash?version=nightly&filters=includePrereleases&target=pocket
2. Flash your RM Pocket with that firmware - either directly through a Chrome based web browser using DFU or via the bootloader. c.f. https://edgetx.gitbook.io/edgetx-user-manual/edgetx-user-manual/installing-and-updating-edgetx
3. Enjoy the fact that instant trim once again knows about more than mode 1.

PR #4491 (2023-12-31): fix(radio): Instant trim only works correctly in mode 1
Fixes #4463 

After the ADC refactor PR, instant trim only works in Mode 1. 
Other modes incorrectly map the axis channels when trying to process instant trim.