EdgeTX/edgetx #4466: LUA serialRead() function doesn't work either setSerialBaudrate()

YuriiChornyi, 2023-12-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

### EdgeTx is not reading values from Serial port 1 using UART protocol

[image]

**serialRead**() function was introduced in openTx 2.3.11 and **setSerialBaudrate**() was added in OpenTx 2.3.12
And I don't have issues with reading values on those versions.

Seems like bug was introduced in EdgeTx 2.7 with refactoring in PR #1089.

### Expected Behavior

Values from serial port should be readable inside LUA script (tool, mixer, etc.)

### Steps To Reproduce

My current setup contains raspberry pi zero w, and I was sending values on both 400k and 115200 baudRates, tested on all openTx version from 2.3.11 to 2.3.15.
Issue identified on EdgeTx 2.9.2 custom build with enabled LUA mixer scripts:
1) firmware without chnages tested with 115200 baudRate
2) firmware with modified LUA_DEFAULT_BAUDRATE set to 40000
Will also check with EdgeTx 2.6 and will update here

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

Issue and discussion around olso described in discord thread https://discord.com/channels/839849772864503828/1188228034533273770

raphaelcoeffic, 2024-01-01:
I'm afraid we'll have to revisit this API, as it does not take the multiple serial ports into account. The way it is right now won't work very well.

raphaelcoeffic, 2024-01-01:
@YuriiChornyi It's not really clear to me: is it only `setSerialBaudrate` that doesn't work? As far as I can see, `serialRead()` should work if you assign the serial with "LUA" function (see *Radio -> Hardware Settings*).

pfeerick, 2024-01-03:
@YuriiChornyi If there is still an issue with SerialRead this issue can be re-opened. FWIW, with SerialWrite, once AUX1 was assigned to Lua, it defaulted to 115200 baud, so I would expect SerialRead to be the same, especially given `LUA_DEFAULT_BAUDRATE` defaults to 115200

closed 2024-01-03 by pfeerick

zebbel, 2024-02-04:
Hey,

serialRead() doesn't work for me either.

Neither via USB nor via the AUX ports.

serialWrite() works without any problems.

I have the firmware v2.9.3 on Radiomaster TX16S.

pfeerick, 2024-02-04:
It will work again in 2.10, and with nightlies as of about a month ago.

zebbel, 2024-02-05:
I have flashed the nightlie build from today.

Unfortunately it still does not work.

I have tried it via USB and via both AUX inputs.

Sending works without problems but I can't receive any data.

I tried serialRead() and sent a string with "\n" at the end and serialRead(1) with "\n" or some other character.

Am I missing something?

The lua script runs as custom mixer. Is it because of that?

pfeerick, 2024-02-06:
No, it seems like something still isn't right. I can repreoduce this on TX16S. Issues is now being tracked here => https://github.com/EdgeTX/edgetx/issues/4605

PR #4501 (2024-01-03): fix(lua): re-implement luaSetSerialBaudrate
Fixes #4466

Summary of changes:
- added `serialGetBaudrate`, `serialSetBaudrate` and `serialGetModePort`,
- re-implement `luaSetSerialBaudrate`.