EdgeTX/edgetx #4467: MT12 MIXERS MONITOR issue

chofchop, 2023-12-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After starting MT12, the display on the mixers monitor screen is incorrect.

https://www.dropbox.com/scl/fi/m8p30or93bgvzw99hqj9j/VID_20231226_201326.mp4?rlkey=6siq120uyq0rmyvnn271tavyf&dl=0

Once, go to the setup screen, it will be displayed correctly.

### Expected Behavior

It should display correctly even after startup.

### Steps To Reproduce

Go to the setup screen.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Anything else?

Firmware : mt12-fbe06b3.bin

chofchop, 2023-12-26:
The video has been replaced.

https://www.dropbox.com/scl/fi/7tsznhjy40g3v1xtrtbrd/VID_20231226_204925.mp4?rlkey=mdy0kkza0whx4bwlhcvog3kv7&dl=0

3djc, 2023-12-26:
Are you talking about « inv » notifications ? Because one time you are in mixer monitor (where they are rightfully not displayed, another in output monitor (where they are)

chofchop, 2023-12-26:
Sorry, I couldn't distinguish between mixers monitor and channels monitor.

However, is it normal for MT12 to forcibly switch to mixers monitor after startup? X9 Lite (ETX 2.8.1) becomes a channels monitor after startup.

3djc, 2023-12-26:
Maybe a 2.10 thing, cannot remember

chofchop, 2023-12-26:
It seems strange after all.

After startup, it will forcibly switch to mixer monitor, but in this state it will not switch to channel monitor even if press the rotary button.

Once I enter the setup screen and switch to channel monitor, I can use the rotary button to switch between mixer monitor and channel monitor.

3djc, 2023-12-27:
Yes the navigation is different in main view than it is when using the shortcut. I'll have a look at the possibility of making the closer, but not sure it will possible

PR #4490 (2024-01-02): fix(bw): Channel Monitor view not working correctly on startup
Fixes #4467 

If the 'Channels Monitor' is the start view (or is selected after startup), the view opens on the 'Mixers Monitor' instead and the ENTER key cannot be used to change view. 

User must enter a setup menu and return to the main view for the channel monitor to work correctly.

On startup - the view is not initialised correctly.
closed 2024-01-02 by pfeerick