EdgeTX/edgetx #4468: Telemetry voice message: Type of value missing

boomerschnecke, 2023-12-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Within the current telemetry voice message its "only" mentioned the value and the coresponding unit.
so f.e. 7 Volts, 4 Amps, 399milliAmperhours ect
Missing is to name beforehand the "type of value"

### Describe the solution you'd like

Missing is to name beforehand the "type of value"
f.e. **Voltage**: 7 Volts, **Current**: 4 Amps, **Capacity**: 300 mAh ect
So the type of value is missing currently (**Volltage, Current, Capacity**). (if possible of course for all telemetry items)
I think this would be a highly appreciated feature update

### Describe alternatives you've considered

no alternative

3djc, 2023-12-27:
I don't think it should be added, since the units themselves indicate what the sensor is about.

You can easily create those sounds and prefix the announcement by this if you so desire

boomerschnecke, 2023-12-27:
Some of them yes, but f.e. Sats: its only a number without anything. If you want to distinguish between ReceiverBatVoltage and normal supply Bat Voltage?. I personly would prefere this

3djc, 2023-12-27:
Well in your example, both voltage would say ´ voltage ´, so it would not help.

boomerschnecke, 2023-12-27:
The intention should be not only voltage, but Battery Voltage or Receiver Voltage or Number of Satelites ect
Jeti Telemetry f.e. is able to do a lot of this things

3djc, 2023-12-27:
The issue here is that sensors come in all kind of shapes, forms and protocols here, unlike Jeti closed environments. There is no way to know where a voltage sensor comes from