EdgeTX/edgetx #4469: ZORRO ELRS - Default Channel Order ignored in new mixer page

andrewnewton, 2023-12-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Default channel order is set as AETR but new blank mixer page entries appear in RAET order.
No entries in the Inputs page, starting with a blank Mixer page and manually entering new lines.
[image]
[image]
[image]
[image]

### Expected Behavior

Mixer lines should appear in AETR order

### Steps To Reproduce

Delete all INPUT entries and all Mixer lines.
Manually enter mixer lines

### Version

2.9.2

### Transmitter

RadioMaster Zorro

### Operating System (OS)

macOS

### OS Version

Sonoma 14.1.2

3djc, 2023-12-27:
No, that's not how it should work. Default channel order is used only for NEWLY created model. Current and existing models will not be affected.

Create a new model, it should be AETR

andrewnewton, 2024-01-02:
No this is changed from 2.8 
With 2.8 I create a new model
Inputs and Mixer are pre populated in AETR order
Then I delete the pre entered Inputs and Mixer lines
Then when re entering Mixer lines, they appear in AETR order
BUT in 2.9 the reentered mixer lines appear in RAET order

pfeerick, 2024-01-02:
I can't reproduce this behaviour. On 2.8.4 and 2.9.2 (and a recent 2.10 nightly), the order the first four channels are added on the inputs page is unchanged - it follows the default channel order setting for RAET, TAER and AETR. The mixes page simply adds the same index input (and always has from the looks of the 2.8 code), so is unaffected by the default channel order settings. Since changing the default channel order setting, and then deleting the inputs and readding them changes inputs 1 through 4, the stick assigned to each input will be different, so the mixer line **should** be the unchanged anyway.

pfeerick, 2024-01-02:
Actually, I just noticed you said with a *blank* inputs page (which is highly unusual btw)... and that seems the be the important factor here:

RETA order on 2.9.2 gives (yes, a double rudder bug there also it seems)
[image]

AETR order on 2.9.2 gives
[image]

RETA order on 2.8.4 gives
[image]

AETR order on 2.8.4 gives
[image]

andrewnewton, 2024-01-02:
Thanks for checking this. Here's a video comparing 2.8 with 2.9.2 https://youtu.be/poGl0lLqzVw