EdgeTX/edgetx #4470: horus x12s with edgetx 2.9.2

grafwiloff, 2023-12-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

hi, yi cannot calibrate the little red joystick … :-(
gr Matthias

### Expected Behavior

it has to be calibrated …

### Steps To Reproduce

i miss it on the calibration issue …

### Version

2.9.2

### Transmitter

FrSky X12

### Operating System (OS)

macOS

Luznatural, 2023-12-27:
Repeated. Solving this problem is already on issues #535 Add calibration of analog joystick and #2361 Calibration screen improvements.

pfeerick, 2024-03-04:
Duplicate of https://github.com/EdgeTX/edgetx/issues/535

closed 2024-03-04 by pfeerick