EdgeTX/edgetx #4479: USB Trainer not find on 2.9.2 French version

pierrotm777, 2023-12-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hello,
I have found an issue for use the trainer mode with the AUX 1 with the French Heli version.
The USB Trainer mode isn't exist into the list.
This issue isn't exist with normal 2.9.2 version.
And the AUX2 don't exist on the setting screen. 

### Expected Behavior

Give a French version with USB Trainer mode

### Steps To Reproduce

Update from 2.8.4 to 2.9.2 French Heli version 

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

W11

### Anything else?

...

3djc, 2023-12-30:
You mean SBUS ? Seems there to me

[image]

pierrotm777, 2023-12-30:
Yes

3djc, 2023-12-30:
Well, most definitively here on pfeerick language builds

[image]

[image]

pierrotm777, 2023-12-30:
Ok, i shall try my firmware upload from 2.8.4 to 2.9.2 fir see if i have again the same issue asap :-)
Thanks for your tests