EdgeTX/edgetx #4483: Horus X12S Switch Bug

ammikey, 2023-12-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Switch ID is lost in the programing screens ie: timer activation switch choice shows only arrows without the switch designation such as SH, or SG. Went back to 2.8 and it works as expected. 

### Expected Behavior

You should be able to see switch designations in drop down boxes.

### Steps To Reproduce

Try to set switch actions no designations found just arrows.

### Version

2.9.2

### Transmitter

FrSky X12

### Operating System (OS)

Windows

pfeerick, 2023-12-31:
I'm not seeing this on the Companion simulator with a new setup, so perhaps check what Radio Settings -> Hardware -> Switches shows for SG/SH ... what is the name of the switch set to there? It should be showing `---`. There was a quirk in the conversions at one point where it was actually set to `---` rather than being an empty value... so perhaps also clear the field by backspacing in it just to be sure.

ammikey, 2024-01-17:
Hi Peter. I just flashed the transmitter again with 2.9 and I still get the same problems the switch Id's are not in the drop downs and the input calibrations are way off. I reflashed 2.8 and everything works fine. I tried my transmitter setup in the 2.9 simulator and it works good there but not in the transmitter.

TramALot, 2024-01-17:
I have gotten some of weird issues now and then. With input names not
wanting to change after I had changed them once before. And it usually
comes down to I left a global function or a special function set up some
place and for some reason that causes it to hold/hide the name(s). But if
you delete everything and start fresh it seems to go away(making a new
model still fails). That's just my observation to something like what
you're describing.
And with a new radio like my T20 from jumper. The channel map sometimes
needs to be manually adjusted on the inputs after switching in ms.

Hartinger, 2024-01-20:
> I'm not seeing this on the Companion simulator with a new setup, so perhaps check what Radio Settings -> Hardware -> Switches shows for SG/SH ... what is the name of the switch set to there? It [...]

I had the same problem with my Boxer and found indeed that the names of all controls consisted of three blanks. Removing these blanks cured the problem.

pfeerick, 2024-01-20:
> I have gotten some of weird issues now and then. With input names not wanting to change after I had changed them once before. And it usually comes down to I left a global function or a special [...]

This may be a different bug, which was fixed in 2.9.3

> The channel map sometimes needs to be manually adjusted on the inputs after switching in ms.

If you mean the Def chan order (Default Channel Order)... that really only applies to creating new models, or if you delete all the inputs and re-add them. It won't change the order of existing inputs for models.

pfeerick, 2024-03-04:
@ammikey I need you to check the `RADIO/radio.yml` file as per https://github.com/EdgeTX/edgetx/issues/4483#issuecomment-1873049090 or upload it here (either zip your MODEL/RADIO folder up and upload it, or change the extension of the file to txt to upload it to github) order to progress with this any further.

ammikey, 2024-03-05:
> RADIO/radio.yml file as per #4483 (comment)

I'll do it in a couple days

pfeerick, 2024-04-08:
few days plus reminder ;)

ammikey, 2024-04-08:
Hi Peter, I'll get that yml file sent you tomorrow. Sorry it's taken so long.
Thanks Mike J

closed 2024-08-04 by github-actions[bot]