EdgeTX/edgetx #4487: Editing "Custom Screen" in model setup - is it possible

pnedic, 2023-12-30:
Hello

Is it possible to edit fields and boxes in "custom screen" tab on model setup page? 
Currently, on that screen it is just visible that some fields are populated, but click or double click on some box does nothing.
Custom screen has to be edited only on TX which is quite big issue when you upgrade your radio from OpenTX to EdgeTX.

Or this is currently impossible (with version 2.9.2) ?

pfeerick, 2023-12-31:
Is this in relation to a color lcd screen radio - as you also never mentioned what radio this was in relation to? If so, then no, it is *NOT* currently possible to configure the custom screens / main views in Companion - it is a feature that has not yet been completed. However, if you have general layout that you want to use on multiple models, it is possible to open the `.yml` files in a text editor, and copy that to your other models... have a look for `screenData` for the main views and `topbarData` for the top bar widgets.

pfeerick, 2023-12-31:
Duplicate of #3686

closed 2023-12-31 by pfeerick

pnedic, 2024-01-02:
Thanks @pfeerick, I'll check for .yml files. I am using Jumper T16 and latest EdgeTX.

pnedic, 2024-01-04:
Yes, it is possible to edit .yml file. There is section "screenData" in .yml file that can be copied to desired modelX.yml file. There should be slightly changes to be done (e.g. model name). Copy (overwrite) edited modelX.yml to SD card MODEL folder and this works perfectly.