EdgeTX/edgetx #4493: OTA for newer Archer receivers doesn't work

ParkerEde, 2023-12-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I try to flash a new ARCHER PLUS Rx via OTA, I get the error "Firmware Updateerror - Transfer failed" right at the beginning.
Old ARCHER Rx can be flashed normally via OTA.

### Expected Behavior

ARCHER PLUS RX should be updateable via OTA

### Steps To Reproduce

start regular update procedure via int.OTA with ARCHER PLUS Rx

### Version

2.9.2

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Windows

### Anything else?

it is the same issue with current main

BompaBert, 2024-01-04:
> ### Is there an existing issue for this problem?
> * [x]  I have searched the existing issues
>
> ### What part of EdgeTX is the focus of this bug?
> Transmitter firmware
>
> ### Current Behavior
> When I try [...]

I have the same setup and OTA works fine. Am assuming the receiver has been successfully registered and bind. If not, be sure you have the latest ISRM firmware installed. This should be version 2.1.10 for SRx+ and lastest receiver firmware 2.1.14.

ParkerEde, 2024-01-04:
Yes, it has been successfully registered and bound. ArcherSR8PLUS is found correctly.
ISRM is 2.1.9EU and SR8plus is 1.0.10.
Start the flash process by pressing Enter in the file /FIRMWARE/AP_8CH_1.0.10.frsk, select "Flash RX by int. OTA", turn on RX, select RX SR8PLUS, accept message window: Current version 1.0.10 [NO] [YES] with yes and then the error appears:
Firmware update error
Transmission failed

ParkerEde, 2024-01-04:
it's  the same with QX7ACCESS

BompaBert, 2024-01-05:
Looks like your SR8 has the latest firmware -- "Current version 1.0.10".

ParkerEde, 2024-01-05:
Yes, the Rx already has the latest version. But flashing the already installed firmware must still work and it does with the old Archer R8 pro.
In the meantime I have also noticed that flashing via ETHOS does not work with the Archer SR8 plus either. I also have the latest ISRM firmware 2.1.10 on it. I assume that it is not an EdgeTx problem, but an ISRM firmware problem

pfeerick, 2024-01-05:
It could be that, or because Ethos is heavily inspired by OTXs design, a
common bug.

BompaBert, 2024-01-05:
Yup, agreed. As I do not have any issues with flashing / re-flashing the
SR10+.

ParkerEde, 2024-01-24:
Andreas Engel (distibutor for Germany) has opened a case on Frsky. As soon as there is feedback, I'll let you know here.

ParkerEde, 2024-01-30:
Hi @BompaBert ,
Frsky is trying to recreate the problem and it's working for them so far. I assume you are using FCC. And I assume Frsky is also testing this issue via FCC. We have LBT in Germany and Andreas Engel and I can immediately reproduce the problem.

BompaBert, 2024-01-30:
Yes, Am out of the US - confirmed. Hope this helps.

ParkerEde, 2024-07-31:
This problem is specific to the Archer plus SR8 in combination with Horus X10 Express and Taranis QX7 ACCESS. All other Archer plus radios work with these radios without any problems. Since the problem also occurs with Ethos, it is quite clear that it is due to the RF firmware in the ISRM of the X10 Express or QX7 ACCESS, since other radios get AP SR8 OTA flashed. Andreas Engel is informed, he can also adjust it. I am out of here, as I no longer have a FrSky radio to test anything.

closed 2024-07-31 by ParkerEde

pfeerick, 2024-07-31:
Thanks for the update!