EdgeTX/edgetx #4494: corrupt luac file after first run

ParkerEde, 2023-12-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

ETX generates a corrupt luac file for the official OTX LUA configuration script for the ARCHER PLUS SR8. This must therefore be deleted manually before each start so that the script can be restarted.
This is about this file:
AP Stab Rx_OPTX.lua
see atached ZIP.
Its reproduceable without Rx.

### Expected Behavior

luac file should be consistent

### Steps To Reproduce

1. start attached LUA
2. stop with RTN long press
3. start LUA again
4. observe there is an error popup
5. delete corresponding luac
6. start lua again
7. observe, now it is starting
8. and so on

### Version

2.9.2

### Transmitter

FrSky X10 Express / X10S Express (ACCESS), RadioMaster TX16S / TX16SMK2

### Anything else?

it's the same with current main

philmoz, 2023-12-31:
2.9.2 works on my TX16S - all Lua scripts compile correctly to .luac files and can be run from them.

2.10 fails on all Lua script compilation - the resulting .luac file is only 512 bytes long.
Running debug mode on 2.10 I see a 'SD WriteBlocks failed' error message when trying to write the .luac file to the SD card.

philmoz, 2023-12-31:
PR #3957 appears to be the culprit. The commit before this works fine on my TX16S.

Flashing the commit for 3957 causes the Lua compile to fail on saving the .luac file; however it is not 100% - some scripts will still compile and save on my TX6S.

ParkerEde, 2023-12-31:
hey @philmoz . Happy new year🤙

philmoz, 2023-12-31:
Happy new year to you also.

raphaelcoeffic, 2024-01-01:
@philmoz @ParkerEde would you mind testing https://github.com/EdgeTX/edgetx/pull/4500? As it seems, the code path without DMA-compatible buffer was not quite tested properly (`luaDumpState` uses a file buffer allocated on the stack).

philmoz, 2024-01-01:
TX16S and EL18 both work with this fix.

ParkerEde, 2024-01-01:
@raphaelcoeffic tested with X10Express and TX16S. Both okay

PR #4500 (2024-01-01): fix(sdio): Check card state after each block written
Fixes #4494
closed 2024-01-01 by pfeerick