EdgeTX/edgetx #4497: Wireless Model Share

jmxp69, 2023-12-31:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Looking for a method to wirelessly transfer a model.

### Describe the solution you'd like

I'm aware this is a wishlist type feature as I'm not sure even ELRS BT has a file transfer capability and I don't know what the roadmap is for 802.11x, but the request is simple:

Wireless model/image/log transfer from one radio to another. Ideally there could be an Airdrop type functionality so pre-approved friends show up as a destination without any pairing requirement.

Since I have a wireless wishlist started, a few other ideas to extend functionality:

- Wireless backup to PC
- Wireless dropbox for firmware/image uploads to radio
- Wireless sync to companion

### Describe alternatives you've considered

Bungee jumping while shooting tequila

### Additional context

Thanks guys. I know this is out there, but just a couple of ideas for the future...

pfeerick, 2023-12-31:
This is basically impossible at the present time... other than possibly with FrSky transmitters (for the handsets with with built in bluetooth), there is no mechanism in which this can be carried out. And for this to happen with just one protocol (i.e. ELRS), would require that protocol to have support for this use case - which ELRS currently does not. This is something that we have discussed internally as a want... but it's a chicken and the egg scenario... the hardware needs to exist for this before the software can implement anything.

jmxp69, 2023-12-31:
Appreciate the comment. I figured as much with current hardware, but even elrs rx’s have WiFi transfer capabilities so if WiFi were possible somehow these are some features I’d hope to see.

pfeerick, 2023-12-31:
The problem therein is this is only available to ELRS... nothing is exposed to the transmitter. What is more likely to happen is for someone to use the AUX ports on a transmitter (well, two, obviously) , attach a ESP826x or ESP32 (i.e. the same WiFi MCU ELRS are using) and then figure out some communications protocol and UI to go with it for ETX... but given the lack of progress recently on the non-frsky bluetooth API ... don't hold your breath as this is an even steeper hill to climb! 😆

jmxp69, 2023-12-31:
Oh...I'm holding it now.

rotorman, 2024-01-01:
The following project looks promising in this direction, but has been stalled for some time now: https://btwifimod.gitbook.io/ and https://github.com/dlktdr/BTWifiModule

Luznatural, 2024-03-24:
Frsky Ethos 1.5.0 already is able to send/receive models, images and files between radios using Bluetooth feature. Could be possibly try to figure out something similar for EdgeTX?