EdgeTX/edgetx #4504: Throttle trim handling in surface radio

3djc, 2024-01-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Throttle trim applies on the whole range, and it's max value is at -1024 instead of 0

### Expected Behavior

Throttle trim only applies to forward section

### Steps To Reproduce

On default model, activate throttle trim
At full reverse, you cannot reach -100, and 0 is reach only after some reverse push

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

pfeerick, 2024-01-03:
On the car radios I have, the trim for steering (ch2) starts off a 0 (neutral), and trims 30 units forwards and backwards. It applies all the time, and can allow you to increase the throttle slightly if made more positive, or increase the reverse if lowered. i.e it allows for the throttle to zero properly.

3djc, 2024-01-04:
That doesn't need trim throttle to be activated.

I have a radio setup here with it activated using proposed scenario, for a car where you have a servo for engine throttle, and other servo(s) for brake
- using thrim throttle you only adjust the low part of engine servo (your low rpm point)
- brake channel isn't affected (and can use another trim if you whish)

mshagg, 2024-01-05:
Most gas and nitro cars have throttle and brake on the same servo, so the behaviour described is expected and unavoidable. 

Using trim to set idle speed would be considered bad practice, except for startup where "idle up" setting is used until normal operating temperature is reached (easy done with a special function to advance throttle slightly).   The centrifugal clutches are engaged by engine RPM, so there is no situation where throttle and brake are applied at the same time (and hence the need to have idle set correctly on the carburettor).

For setups using seperate brake/throttle servos, this would need to be mixed accordingly by the user anyway, and trim settings on the throttle channel wouldnt impact the braking channel.

3djc, 2024-01-05:
That's fine, the behaviour pfeerick was describing is the one by default. There is another one that was completely broken when throttle idle mode is activated . I'm trying to make it sensible at least, even if not widely used (it will work very nicely for those with separate throttle /brake servo)

pfeerick, 2024-01-05:
So you meant the throttle [idle] trim, not throttle trim? 

In that case, for nitro, and also some crawlers, yeah, another option of of that would be nice . It may be a good idea to **not** have it restricted to surface radios though, as you can have a centre sprung throttle on a conventional transmitter also.

3djc, 2024-01-05:
Yes yes, the exact name on radio is T-Trim-Idle, sorry if I have not been clear from start.

Do we have a standard way of declaring a centre sprung throttle on a conventional transmitter ? Because a lot of things won't work properly for those imho

pfeerick, 2024-01-05:
No. What is there really to go wrong though? Throttle timer triggers won't work, and probably the throttle idle trims? The throttle position for pre-flight checks can be defined now, which was contributed by a user with that exact setup.

3djc, 2024-01-05:
At least those 2 yes, I would have to scroll through the changes I had to make for MT12 to have an exact list. I guess a proper fix for that setup should be done

pfeerick, 2024-01-05:
Hm... in that case... can I suggest (again) that is be a great reason to have a config value for the throttle time trigger value... allows you to tune it for overly-sensitive low point on some transmitters (was requested by a flying buddy), and also set it to a custom value for centre-sprung throttle. There are more than a few applications that aren't just surface (boats, cars, altitude-hold multirotor, etc.). Don't think it needs to be anything more than radio level setting... until someone comes up with that exact use case/need... 🤪

3djc, 2024-01-05:
I think that would be a bad response: your overly sensitive transmitter simply needs to be gently calibrated (people tend to push hard during calibration, and that triggers those behavior). If throttle is not at 100% because of calibration, there is a real risk that a propeller, blade, you name it, will start.

Centre-sprung throttle should not need a custom throttle check value, it should simply be declared somewhere (either compile or runtime), and settings adjust accordingly, since more than throttle start needs to be checked (and for those, throttle check could be disable, since the spring ensures stick in the proper position anyway).  But that probably for a different PR

pfeerick, 2024-01-05:
While you are indeed correct that people tend to push hard on calibration. some hall gimbals are more pone to a couple of % drift (I have T-Lite that is particularly bad... even after gentle calibration, have to push the stick hard to bypass the pre-start checks)... which is annoying since the motors doesn't actually engage until it is above 5.5% from idle anyway (FC safety checks). And yes, that is material for discussion and a different PR if anything eventuates. 

> (and for those, throttle check could be disable, since the spring ensures stick in the proper position anyway)

Don't assume... it should still do the checks... as what if the stick is pushed over or the spring snapped, etc.

3djc, 2024-01-06:
While working on radio test for surface, I noticed something wrong. In fact, in current implementation the trim potency is doubled compared to air radio for throttle channel (regardless of trim idle or not), this is because full trim result in the same +25 value, but when +25 is 1/8 of the total range for air radio (-100 to 100), it is 1/4th of the forward (or reverse) channel. I have therefore introduced a division of trim value for the th channel o surface radio

PR #4520 (2024-01-21): fix: enable radio test for surface radio
WARNING: Include mixer fixes, please review thoroughly

This fixes #4504
closed 2024-01-21 by pfeerick