EdgeTX/edgetx #4508: Erratic GPS Distance values in Telemetry

BompaBert, 2024-01-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

A static (non moving) Frsky GPS module shows a correct GPS location and Altitude however distance calculation readings varies from 0 - 60000m what makes it unusable.

### Expected Behavior

The Distance values are correct with the same GPS module and setup using OpenTX.

### Steps To Reproduce

1) Connect an Frsky GPS Module using an S-Port or FBUS.
2) Discover the new sensor in Telemetry.
3) Obtain a static valid GPS location and Altitude in the telemetry view.
3) Add a new sensor using Calculated / Distance with GPS and GPS Altitude.

### Version

2.9.2

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

[image]
[image]

frankiearzu, 2024-01-09:
Will compare the code to see what has changed between OTX/ETX.

pfeerick, 2024-01-09:
In that case, @BompaBert you should be able to try the firmware for your transmitter from this file https://github.com/EdgeTX/edgetx/actions/runs/7429943710/artifacts/1151445870 (it's the latest build from the 2.9 branch, which includes that fix. It should be safe to use regularly as it is pretty much the 2.9.2 release + a couple of minor bug fixes, and is basically what will be released as 2.9.3 soon.

pfeerick, 2024-01-09:
Tentatively marking as 

Duplicate of #4437

BompaBert, 2024-01-09:
Flashed with Firmware x10-access-22673b5.bin. Same results see attached.
Log collected with 0.1 sec interval.

BompaBert, 2024-01-10:
My bad. Ran the 22673b5 again but this time via bootloader method. All ok now.
Thanks for jumping on this so quick. I'll close the issue.

closed 2024-01-10 by BompaBert

mha1, 2024-01-10:
You're welcome

qsiguy, 2024-01-13:
I'm working with a Spektrum SkyID module testing telemetry trying to generate a calculated "height from takeoff" and I only have GPS coordinates and current altitude.  Frankie suggested I try the fix above.  I flashed the tx16s-22673b5.bin firmware and generated the attached telemetry item but all the new "Dist" value shows is the altitude.  No distance is calculated.  What should it be doing?
[image]

mha1, 2024-01-13:
Is a GPS connected and live? Do have a log? Does the GPS position change? If it doesn't distance = altitude

qsiguy, 2024-01-14:
No log at the moment.  Yes, it is a live, real time GPS and the location data does change as I walk around.  I just walked with it about 50 yards back and forth, raised and lowed the unit about 8-9 ft to see if I got any response. It just always showed current altitude.  Didn't fly with it yet.  Early in my testing of the telemetry capabilities of the unit.