EdgeTX/edgetx #4514: 3 position sticky trim

Kevltan, 2024-01-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I would like to see added a "3 position sticky trim". This would be able to used in all flight modes including fm0.

### Describe the solution you'd like

As above

### Describe alternatives you've considered

I have considered other options but this must be shown on the TX display as normal trim movement. It may have other uses but in my case it would be used for the throttle trim - trim min, trim centre and trim max. So from trim min, 1 click up trim centre, next click up trim max and the opposite would apply. There could also be an option for a "2 position sticky trim" being added - trim min and trim max etc.

### Additional context

N/A

pfeerick, 2024-01-08:
Just to keep the information for this together in one place, it seems like it should be an extra trim step mode... i.e. coarse is 32 steps vs 128 steps for fine, whereas this would just be three.  

The manual snippet that was shared from the Spektrum DX18 manual explains that it can be used to set the throttle trim low, middle or high:
[image]

philmoz, 2024-01-08:
That would apply to all trims as we only have one trim step setting (at the moment).

pfeerick, 2024-01-08:
Indeed. 

> Although, the catch is that changing the trim steps changes this for all of the trim controls... so maybe something some special handling along the same lines of the 3P functionality would be needed. Although Phil is right... using some SF Lua logic the manipulate the trim would nicely circumvent the need to do anything more to the trim code.

Kevltan, 2024-01-08:
> Just to keep the information for this together in one place, it seems like it should be an extra trim step mode... i.e. coarse is 32 steps vs 128 steps for fine, whereas this would just be three.
> 
> The manual snippet that was shared from the Spektrum DX18 manual explaining the purpose of this type of feature is that it can be used to set the throttle trim low, middle or high, useful for turbine models: ![image](https://private-user-images.githubusercontent.com/5500713/294823827-3addde32-9cf9-4888-a6ed-490bde3ce682.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ2OTcxMDUsIm5iZiI6MTcwNDY5NjgwNSwicGF0aCI6Ii81NTAwNzEzLzI5NDgyMzgyNy0zYWRkZGUzMi05Y2Y5LTQ4ODgtYTZlZC00OTBiZGUzY2U2ODIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDhUMDY1MzI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGIxYTdjYzQyMzQxMDRlYmM5MDM3YWE0NjFlY2VjZTIwZTAyODcyNjFiYWExMzI0ODk0MGZjNTAyODNhYmUwNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.9kC6sDoXpaK5hcgq_VTT2BED-UrtylcFW2sTwUOrJwo)

Would it not be just 2 steps?

Kevltan, 2024-01-08:
But you are already in one of the steps so if you are in trim low you need 2 presses of the trim up to get to high. If you are in trim centre/mid position you need one press up or down on the trim to get to high or low. So whatever position the trim is in you need maximum of 2 steps yes/no?

Kevltan, 2024-01-08:
It's all according what you mean by "steps" there are 3 positions but only 1 or 2 steps either way are needed to achieve one of the 3 positions because you are already in position 1, 2 or 3. Yes/no?

pfeerick, 2024-01-08:
I was referring to the maximim number of trim "positions", as anything else
is relative to the current position, so is just confusing.

Kevltan, 2024-01-10:
https://github.com/EdgeTX/edgetx/assets/56512603/3f912fd9-d6fe-4085-929e-4d2ccf16d39c

https://github.com/EdgeTX/edgetx/assets/56512603/7ee17761-7c29-4561-bb00-2aecae7ba022

Kevltan, 2024-01-10:
I have changed Heading title and text to better reflect the enhancement asked for.

Kevltan, 2024-01-10:
> Spektrum DX18

Spektrum DX18QQ

Kevltan, 2024-01-10:
https://youtube.com/shorts/709BjZkYfxs

Kevltan, 2024-01-10:
@philmoz I am sure this part of the idea you were going tell me earlier on but I think now a picture/ video paints a thousand words/ texts!!!

Kevltan, 2024-01-11:
@pfeerick Hi Peter, please take a look at this and let me know what you think. Is there an easy way to link this v = # to an option in models or a special function not gvars? Would it be easy to do?

Kevltan, 2024-01-17:
[image]
Thanks for help with code @mha1👍

philmoz, 2024-01-27:
Haven't forgotten about this; just trying to work out a suitable (general purpose) solution.
I don't think using a GV is a good idea.

3djc, 2024-01-27:
Isn't that much easier to do user side now that we have 3P support on trims ?

pfeerick, 2024-01-27:
By definition, 3P disables trims, no? The functionality he wants is a three position trim... https://github.com/EdgeTX/edgetx/issues/4514#issuecomment-1880442429 ... if we had separate trim coarseness settings that could just be another trim step setting... but since we don't... maybe another "Throttle" specific option that can manipulate the trim steps for the throttle?

Kevltan, 2024-01-27:
I have found the above works fine with the advantage that it is completely customisable.

mha1, 2024-01-27:
using a GV was never intended to go into main. This was just a code snippet for @Kevltan's proof of concept. I'd say this needs a separate yaml variable and some UI in the throttle settings

Kevltan, 2024-01-27:
@mha1 Absolutely👍

Kevltan, 2024-01-27:
Hi @3djc, its more than Spektrum because if its done right it is a fully customisable as I have proved above and I am sure with the devs input it would be polished to perfection. For your information I have never used a 3 position trim for any setup I have ever had other than with the edit I have made recently to EdgeTX. I did not come up with the idea from Spektrum I just happened to be looking for options for this for my setup and found it online (didnt realise my disused DX18QQ could do this). I thought it would be a good idea - an enhancement to EdgeTX, but unfortunately I appear to be one of the "handful of users"!
But isn't this what EdgeTX is all about being better than the "others" and being able to do just about everything imaginable?
If you feel this is not the way to go then thats fine, the edit I made works for me and it means I just a have to change a few lines every time I update the firmware in EdgeTX.

I am beginning to wish now that I hadn't used Spektrum as a comparison - Spektrum is Sh%$*. I have a Spektrum DX18QQ that I regret ever buying and in my opinion the worst radio I have ever bought, but remember Spektrum radios have firmware, software and fly models too so they all have similar "behaviour".....My bad!!

philmoz, 2024-01-27:
@Kevltan I think you mentioned at one point this was for handling the startup sequence for a turbine engine.

You need three easily reproducible throttle settings (STOP, IDLE and MAX) to program the ECU and the initiate the startup sequence.

Having done a bit of research I believe the idea of using trim to do this dates back to the days of mechanical trims and limited programability of radios.

IMO you would be better to use a throttle cut switch and configuration of the throttle input/mix to do this. E.G. set weight to 90 and offset to 10 in the throttle input - this gives you IDLE (stick down) as 10% throttle and MAX (stick up) as 100%. Set a throttle cut switch to override throttle to 0% and you have your STOP setting.

Kevltan, 2024-01-27:
@philmoz. You are correct except anything below mid trim is turbine cut anything above idle setting is turbine run so a 2 position trim would do it really. But for safety reasons of safety 3 position or more is best and gives access to being able to cut the turbine quickly if necessary. You will also be aware that I use a widget screen with this that has an idle trim only. I use a flight controller for the helicopter so no other trims are necessary and are disabled and used for for arming only etc. As shown above I have found a way of doing this and thought this might be useful other users. I did not think this would be such a hard thing to implement into EdgeTx as an option but I seem to be finding a lot of opposition to it here!

philmoz, 2024-01-27:
It's not a question of being hard to implement or being opposed to solving this.
We have a lot of legacy 'stuff' that has crept in over the years because someone thought it would be a good idea. 
A lot of it is poorly documented and hard to understand which makes life difficult when trying to make changes.
The more the problem is understood the better the final solution can be for everyone.

What is the widget you are using? Can it be changed to be more flexible?

Kevltan, 2024-01-27:
@philmoz 
https://github.com/EdgeTX/edgetx/assets/56512603/eb96b84a-c1e3-428f-970b-7854b178e995

Kevltan, 2024-01-27:
@philmoz 
https://github.com/EdgeTX/edgetx/assets/56512603/95d5631f-44b6-456d-a3b8-9d96b7ba2c1a

Kevltan, 2024-01-27:
@philmoz 
https://github.com/EdgeTX/edgetx/assets/56512603/18b00d50-053f-4505-ae91-7fe8b7688ecb

philmoz, 2024-01-30:
From all the threads I have looked at on rcgroups using a throttle cut switch seems to the preferred and safest option.

What you are doing can be achieved with current EdgeTX logical switches, special functions and a bit of Lua scripting.

Logical Switch:
[image]

Special Functions:
[image]

Lua script (save as SCRIPTS/FUNCTIONS/ttgv.lua)
```
local function init()
end

local TH_TR = 3
local FM = 0
local GV = 8

local function run()
  local t = model.getFlightMode(FM)
  local v = model.getGlobalVariable(GV, FM)
  if (v > 128) then v = 128 elseif (v < -128) then v = -128 end
  model.setGlobalVariable(GV, FM ,v)
  t.trimsValues[TH_TR] = v
  model.setFlightMode(FM, t)
end

return { run=run, background=nil, init=init }
```

Kevltan, 2024-01-30:
Hi @philmoz, thanks for that, looks good but I meant to tell you I have no special functions left. In fact I think I have 3 LS's and 0 SF's I had to lose an SF (play track) to put the constant in! Which is another reason I was looking for other options with a possible screen input option tab in throttle idle only.
 To be honest Phil the firmware edit above to me is much simpler to do and keeps an already complicated model profile clean.
 My edit to the firmware only works if throttle idle setting is selected. But as I said it doesn't quite work as expected  because if SF or GV9 do not have values in it does not seem to default back to #else if v = 4!

Kevltan, 2024-01-30:
> Can we see the list of your SF, we might be able to help simplify it

OK will send all later today. Work at the moment 👍

Kevltan, 2024-01-30:
Model settings:- Requires modified RDT widget and turbine ecu status setting to enable turbine start up using T4 & T1 inwards.  SJ & SI emergency turbine stop T4 & T1 inwards is disabled whilst turbine running. Throttle trim down will also stop turbine. SG timer reset 1 sec after long hold 1st telemetry start and then 2nd turbine start sequence begins. CD timer auto start when throttle advanced. Logic to start up in throttle hold if wanted (RPM > monitor with closed loop but would not be detrimental if tele lost used for throttle hold if auto rotated). T4 & T1 outwards arms and disarms telemetry manualy. Telemetry starts up when turbine start sequence commences and stops when cooled down. Call out of all turbine startup and shutdown sequences and faults. Turbine can be spun up manually with trim back and armed, used to cool if neccessary. Throttle trim must be past mid point to commence startup. Check for ECU battery voltage and connected with warning. Check RX battery voltage with warning etc. etc. etc. All working and tested in flight!

semver: 2.10.0
header: 
   name: "T-Rex 700T"
   modelId: 
      0:
         val: 50
   bitmap: "trex700tt.png"
   labels: "Favorites,Helicopters"
timers: 
   0:
      start: 720
      swtch: "L52"
      value: 168
      mode: THR_START
      countdownBeep: 2
      minuteBeep: 1
      persistent: 0
      countdownStart: -1
      showElapsed: 0
      extraHaptic: 1
      name: "Flight"
   1:
      start: 0
      swtch: "NONE"
      value: 0
      mode: OFF
      countdownBeep: 0
      minuteBeep: 0
      persistent: 0
      countdownStart: -1
      showElapsed: 0
      extraHaptic: 0
      name: ""
   2:
      start: 0
      swtch: "NONE"
      value: 0
      mode: OFF
      countdownBeep: 0
      minuteBeep: 0
      persistent: 0
      countdownStart: -1
      showElapsed: 0
      extraHaptic: 0
      name: ""
telemetryProtocol: 0
thrTrim: 1
noGlobalFunctions: 0
displayTrims: 0
ignoreSensorIds: 0
trimInc: 2
disableThrottleWarning: 0
displayChecklist: 1
extendedLimits: 1
extendedTrims: 0
throttleReversed: 0
enableCustomThrottleWarning: 0
disableTelemetryWarning: 0
showInstanceIds: 0
checklistInteractive: 0
customThrottleWarningPosition: 0
beepANACenter: 4
mixData: 
 -
   weight: 100
   destCh: 0
   srcRaw: I1
   carryTrim: 0
[... 1825 lines cut ...]
         widgetName: "Radio Info"
      5:
         widgetName: "Date Time"
         widgetData: 
            options: 
               0:
                  type: Color
                  value: 
                     color: 0xF8FCF8
view: 0
modelRegistrationID: "634615Q-"
usbJoystickExtMode: 0
usbJoystickIfMode: JOYSTICK
usbJoystickCircularCut: 0
radioThemesDisabled: GLOBAL
radioGFDisabled: GLOBAL
radioTrainerDisabled: GLOBAL
modelHeliDisabled: GLOBAL
modelFMDisabled: OFF
modelCurvesDisabled: GLOBAL
modelGVDisabled: GLOBAL
modelLSDisabled: GLOBAL
modelSFDisabled: GLOBAL
modelCustomScriptsDisabled: GLOBAL
modelTelemetryDisabled: GLO

Kevltan, 2024-02-02:
> By definition, 3P disables trims, no? The functionality he wants is a three position trim... [#4514 (comment)](https://github.com/EdgeTX/edgetx/issues/4514#issuecomment-1880442429) ... if we had separate trim coarseness settings that could just be another trim step setting... but since we don't... maybe another "Throttle" specific option that can manipulate the trim steps for the throttle?

Please look at my firmware edit - it sets "throttle idle" ONLY steps no other trim steps are altered.

Kevltan, 2024-02-02:
> Can we see the list of your SF, we might be able to help simplify it

Have removed SF as not realy needed.

Kevltan, 2024-02-04:
> Haven't forgotten about this; just trying to work out a suitable (general purpose) solution. I don't think using a GV is a good idea.

> Haven't forgotten about this; just trying to work out a suitable (general purpose) solution. I don't think using a GV is a good idea.

Only option available!

Kevltan, 2024-03-07:
> From all the threads I have looked at on rcgroups using a throttle cut switch seems to the preferred and safest option.

Easy to knock a switch accidently or in panic (instant turbine cut not recomended in a heavy turbine heli - trust me I know), not so easy to knock the throttle digital trim.

pfeerick, 2024-03-07:
This I have to agree with 100%... on TX16S I use SF as my arm switch via logic that inhibits both arming and disarming unless certain parameters are met, and instead use T6 up as my emergency instant disarm.

Kevltan, 2024-03-07:
👍

Kevltan, 2024-03-08:
> Isn't that much easier to do user side now that we have 3P support on trims ?
 You are refering to the trim being a 3 position switch NOT a 2 step 3 position trim - min, centre and max. In this config it still acts as a throttle trim, not an asignable switch.