EdgeTX/edgetx #4516: Companion not using selected model when simulation requested

rickalt, 2024-01-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Selecting simulation is not using the selected model

### Expected Behavior

Created a new model, then asked for simulation but another model was simulated

### Steps To Reproduce

1. create new model 2. insure it is selected 3. hit simulation button

### Version

2.9.2

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Win 10 or 11  64bit

elecpower, 2024-01-07:
Are you clicking the Simulate Radio button, as it will load the Default model (bold model) and not the highlighted model.
To simulate the highlighted model, right click it to bring up the context menu and select Simulate Model.
Tested and this loads the highlighted model. Also clicking the Simulate button in the model edit window loads the open model.

pfeerick, 2024-01-07:
Well, to be fair, it says "simulate radio", not "simulate model", so _should_ you really expect it to load the highlighted model (as opposed to the one marked as the default model) when you don't press the "simulate model" button? 

I'm not saying it couldn't be done better... although I'm not sure it really can be in this instance. I think this is just one of those things that need to be learnt like with any new system/software. And yeah, the mixer can be quite daunting, as it is very powerful... although that brings a certain of amount of complexity with it.

closed 2024-01-07 by pfeerick

rickalt, 2024-02-05:
Do you know why the override channel description does not appear in the
drop down menu in special functions on my WIN10 system but does appear on
the WIN11 system. Since the SD card used was the same for both, the
function simulates properly on both computers!

pfeerick, 2024-02-05:
Could you be less specific please? :zany_face: 

Is this with the same version of EdgeTX Companion? Which version is that? And do you mean the special functions tab in Companion, or when simulating?

rickalt, 2024-02-05:
companion 2.9 on both pcs. Special function tab while not simulating.  The
drop down box does not show override channel options on Win 10 but the yml
file contains the override function.  Everything works as expected on Win
11 !!

elecpower, 2024-02-06:
To list or not is controlled by the radio profile build option.

rickalt, 2024-02-06:
it is using the same SD card in both instances. Is the radio profile read
in when I select import from radio?

pfeerick, 2024-02-06:
This option: 
[image]

elecpower, 2024-02-06:
Thanks @pfeerick l was going to do the same later.

rickalt, 2024-02-06:
I'm 87.5% positive that option was not selected. I will let you know. Any
way thanks for your great support. You guys are doing an amazing job. Wish
I could help in some way.

pfeerick, 2024-02-06:
Reporting issues and testing fixes for them helps... not everyone needs to be on the programming side of the "fence" 😉

rickalt, 2024-02-08:
Attached is a screenshot of the xmitter info containing the SD card I used
when I read in the models into Companion 2.9.  I don't see any exclusions
of Override channels!   What firmware is Companion emulating?  I assume it
is what was imported with models.

pfeerick, 2024-02-08:
You'll need to upload that on github... email attachments don't display.

rickalt, 2024-02-08:
[image: edgetx_options.bmp]
Did this work?

pfeerick, 2024-02-08:
No. Only text with come through when you reply via email.

rickalt, 2024-02-08:
Ok, I will.

rickalt, 2024-02-09:
I loaded WIN 10 on a spare computer and fully updated it. I downloaded
Companion 2.9.2 and installed it. I configured a TX16S radio profile. I
synced that SD card with the new PC install. I then loaded the model
showing the symptoms of missing overridech.  It works perfectly.!!   My old
win10 is really old with lots of older SW so I suspect a mixup in windows
libs.  Thanks for your great support

rickalt, 2024-02-17:
This memo is for Neil Horne.  I had the issue with Win 10 working
differently from Win 11.  As a closing comment, I went to the Win 10 system
and created a new radio profile insuring that the nooverride option was NOT
selected. Using this profile Companion reacted normally, so I would
conclude that you were correct in diagnosing that option had been selected.
I'm pretty sure I did not select it with the original radio profile
though.  I was assuming that Companion was reading in the profile from the
SD card which clearly worked properly in the radio. So obviously my
assumption was wrong. Thanks for the help and your patience.