EdgeTX/edgetx #4521: STORAGE WARNING Radio settings invalid *after* first successful boot

yds, 2024-01-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

got my shiny new Jumper T20 and spent a day getting familiar and porting my OneTrueModel-to-rule-them-all on the latest ETX nightly.. EdgeTX is nice and solid on this radio -- props to all the devs for making brand new radios so nice to use right out of the box.. 🤘 my only reproducible niggle is if I take a `radio.yml` file, delete the checksum and bump `manuallyEdited: 1` then copy the file back to the radio -- EdgeTX will boot up fine the first time then when I shut down and power up again ETX will usually reject the `radio.yml` even tho it just booted fine with the same file a moment ago.. this does not happen if I interact with the radio for a while.. I have not found a reliable workaround for the bug other than just keeping it powered on after the first boot and interacting with the radio and setting -- if I power down too soon I get the dreaded `Radio settings invalid` storage warning.. admittedly this is an edge case since most people don't use Vim to tweak `radio.yml` settings -- but some of us like to use Vim for _everything_ 🤓

### Expected Behavior

expected behavior is to _not_ get this
```
   STORAGE WARNING
Radio settings invalid
```
_after_ the radio boots fine the first time.. :wink:

### Steps To Reproduce

steps taken:
- flash the latest nightly
- edit `radio.yml` file, delete the `checksum` and bump `manuallyEdited: 1`
- boot radio and wait for all the voice files to stop playing.. my model is very chatty so it takes a moment for the chatter to die down -- I figure that's long enough for the radio to re-save the `radio.yml` file..
- shutdown and power up again and I get:
```
   STORAGE WARNING
Radio settings invalid
```

### Version

Nightly (Please give date/commit below)

### Transmitter

Jumper T18

### Operating System (OS)

Windows

### OS Version

Win11

### Anything else?

Jumper T20 is missing form the Transmitter pulldown menu in this bug report form 😉

pfeerick, 2024-01-08:
> Jumper T20 is missing form the Transmitter pulldown menu in this bug report form 😉

Fixed. Thanks :)

Can you try again, without deleting the checksum (as that isn't needed - only toggling the manuallyEdited flag so that it skips the checksum). The firmware does wait to write to storage, in order to try and batch writes, but I've not known it to do a half-job (i.e. write the wrong checksum and set manuallyEdited back to 0) ... but maybe that is connected to statup not being completed yet?

pfeerick, 2024-01-10:
There is probably something in the check logic going wrong then. It was never anticipated that you would fiddle with the checksum since that is what the `manuallyEdited` flag was there to invalidate. Did you delete the whole line, or just the checksum value?

yds, 2024-01-10:
> Did you delete the whole line, or just the checksum value?
I was deleting the entire `checksum` line, never just the value. to reproduce all I'd do was delete the `checksum` line, bump `manuallyEdited: 1` and save the file.. and more often than not without the checksum line the radio would boot fine the first time, but then the second boot would cause that `Radio settings invalid` storage warning.. and eventually it would just work, sometimes it would break even if I played with the radio for a while tweaking both model and radio settings and the second boot would still fail.. however if I leave the `checksum` line alone and only bump `manuallyEdited: 1` I could not reproduce the fail even once..

I was expecting that when `manuallyEdited: 1` is set, the `radio.yml` file would get all fixed up and saved by EdgeTX the way it likes it.. but it's acting in weird undeterministic ways when the `checksum` line is missing..

pfeerick, 2024-01-10:
Ok, that means it's taking the code path that assumes you were upgrading from an older version of ETX (i.e. a version prior to the checksum existing)... Thanks... narrows down where to look ;)