EdgeTX/edgetx #4527: Trims switches still make sounds when disabled in DM

txbone66, 2024-01-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Not sure if this is intended bahavior, but this is on the MT12.  I have T1-T4 being used in Special Functions to adjust global variables and T5 as a sticky in logical switches.  All 5 trims made the trim beeps, center, max and min (expected).  When I turned off all 5 trims in DM0, T1-T4 still make the sounds while T5 does not.

### Expected Behavior

I would think that T1-T4 should not make the trim sounds once it's turned off in DM0, T5 is the behavior I'm expecting.

### Steps To Reproduce

Outlined above

### Version

Other (Please specify below)

### Transmitter

RadioMaster MT12

### Anything else?

FW: edgetx-mt12
VERS: RM Factory

pfeerick, 2024-01-09:
Unable to reproduce this with a recent nightly firmware, so this is most likely fixed.

Please note you need to set all five trims to `--` for them all to be disabled (as trims).

philmoz, 2024-01-09:
I think the issue is when a trim is both disabled and used to adjust a GV in a special function.

There is code that overrides the normal trim handling and instead adjusts the GV value. This is what plays the sounds and uses the GV value instead of the normal trim value to determine which sound. It does not look at whether the trim switch is disabled.

txbone66, 2024-01-09:
I was attempting to figure out how to get the sound to stop.  I disabled all the trims in DM0, and noticed that T1 - T4 were still making sounds, and as @philmoz pointed out, even disabled, they are still make sounds with the common thread that they are all in the special function section.  T5 was the interesting one because I'm using a sticky logical switch and then using special functions to overwrite a Channel value based on the !Lx or Lx, and for that one it doesn't make the sound.  T5 is just an on / off for a halogen light bar.

What I'd really like is some control around the sounds, something like this:  
Trim Sounds T1: All / Beeps Only / Center Only / Off

I'm about to head out of town, but when I get back, I'll take a backup and try that nightly to see if I can get it to replicate.