EdgeTX/edgetx #4535: DSM Telemetry: Add RemoteID/FAA sensors to EdgeTX

frankiearzu, 2024-01-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Support for new type of sensor..

Device:
Spektrum SkyID: https://www.horizonhobby.com/product/sky-remote-id-module/SPMA9500.html

The first batch of Spektrum SkyIDs are already out,  SkyID current can send via telemetry the standard GPS data, but also is sending new information. The parts  that we have reverse engineered is FAA-ID and Owner-ID.

### Describe the solution you'd like

Add New Sensor:
FAA (Text): The FAA ID of the device

Same sensor naming can be used once we identify more brands/protocols RemoteID devices.

FYI: Im already started with the coding in EdgeTX

f2knpw, 2024-01-13:
yes @frankiearzu, I have indeed modified my TX16s so that it can broadcast remoteid.
But:
- it's a french remoteid not the FAA one (although I don't exactly know the difference (if any))
- It's my radio which broadcast the Wifi signal (not the drone). This not allowed by the FAA (I have read all the official discussion and the question has been asked...). In France it is not officially written that broadcasting from ground (by Tx) is forbidden (so it's probably allowed !)

Have a look here to see all the details (open source) : https://hackaday.io/project/192752-variometer-gps-tracker-remoteid-for-rc-glider

frankiearzu, 2024-01-13:
If we can send an RemoteID from the Plane/Drone, what do you think of the sensor naming?  RID??

f2knpw, 2024-01-14:
yes why not RiD !

frankiearzu, 2024-01-15:
New Firmware:  TX16S 2.9.3 + SkyID:  https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/edgetx_2.9.3-SkyId.bin

frankiearzu, 2024-06-26:
Just cleanup.. i think is not needed. We can display it with a LUA script.
Unless someone request it to be a proper edgetx sensor to write it to Telemetry logs or display somwerhere, will hold on checking in this.

closed 2024-06-26 by frankiearzu