EdgeTX/edgetx #4544: Taranis Qx7 crash when set input offset to high positive value

GalacticGreg, 2024-01-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If I edit an input, (test with AIL to see this) and change the offset in the positive direction, once I get to about 60, the transmitter crashes and reboots.

### Expected Behavior

I can set the offset without crashing.

### Steps To Reproduce

See above.

### Version

2.9.2

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

### Operating System (OS)

Windows

3djc, 2024-01-16:
Confirmed

philmoz, 2024-01-16:
In drawFunction make this change (in addition to your changes):

```c
  for (int xv = -CURVE_SIDE_WIDTH; xv <= CURVE_SIDE_WIDTH; xv++) {
    coord_t yv = -(fn((xv * RESX) / CURVE_SIDE_WIDTH) * (CURVE_SIDE_WIDTH*2+1) / (RESX*2));
    if (yv < -CURVE_SIDE_WIDTH) yv = -CURVE_SIDE_WIDTH;     <<---- add this line
    if (yv > CURVE_SIDE_WIDTH) yv = CURVE_SIDE_WIDTH;        <<---- add this line
    if ((xv > -CURVE_SIDE_WIDTH) && (abs((int8_t)yv-prev_yv) > 1)) {
      int len = 0;
      if (yv > prev_yv) {
```

I was going to add to your PR; but SourceTree is misbehaving at the moment.

3djc, 2024-01-16:
Well, that produces a probably unwanted horizontal line

[image]

How about:

`  for (int xv = -CURVE_SIDE_WIDTH; xv <= CURVE_SIDE_WIDTH; xv++) {
    coord_t yv = -(fn((xv * RESX) / CURVE_SIDE_WIDTH) * (CURVE_SIDE_WIDTH*2+1) / (RESX*2));
    if (yv < -CURVE_SIDE_WIDTH || yv > CURVE_SIDE_WIDTH) continue;
    if ((xv > -CURVE_SIDE_WIDTH) && (abs((int8_t)yv-prev_yv) > 1)) {`

philmoz, 2024-01-16:
> Well, that produces a probably unwanted horizontal line (unsure, maybe we want the line)

The fix I posted now matches the color LCD curve display.

3djc, 2024-01-16:
matching is good, lets do that then !

GalacticGreg, 2024-01-16:
Thanks for looking into this. Can this be closed?

3djc, 2024-01-16:
It  will be done automagically when PR gets merged

PR #4550 (2024-01-17): fix: strengthen lcd buffer protection
Improve LCD buffer protection by checking negative x/y values

This fixes #4544

NB: 
- this should be applied to 2.9.x too
closed 2024-01-17 by pfeerick