EdgeTX/edgetx #4546: Setting up a volume knob doesn't change start up voice volume

GalacticGreg, 2024-01-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When setting up a volume knob (in global functions "ON. Volume. S1. Enable") the start up "welcome to edgetx" sound is not controlled by the knob.

Further, when I turn the knob completely down during the start-up sound, it skips the splash screen. I also had an issue last night that following the behavior of skipping the splash screen I had a "bad SD card" error but I'm unable to reproduce this issue this morning so it may be another issue.

### Expected Behavior

The volume knob controls all sound volume.

### Steps To Reproduce

See above.

### Version

2.9.2

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

pfeerick, 2024-01-16:
> When setting up a volume knob (in global functions "ON. Volume. S1. Enable") the start up "welcome to edgetx" sound is not controlled by the knob.

This is a known / expected behaviour.  Special / Global functions (as well as a few other things) are not active until after the pre-start checks are done/dismissed.Hence why the volume cannot be set by the knob during startup. Instead, you hsould set your default volume to be low, so that it is low while the radio is starting up, and the knob can take over once you've cleared the checks. This behaviour is expected to change in the future, but we're not there yet. 

> Further, when I turn the knob completely down during the start-up sound, it skips the splash screen. 

You'll also notice the same behaviour when you move some other input... such as the sticks ;) It's supposed to dismiss/skip the splash screen  if you give some input.

GalacticGreg, 2024-01-16:
Ah, thanks for your time, pfeerick.

closed 2024-01-16 by GalacticGreg

naggie, 2024-08-27:
Thanks! Made me realise I had my default volume down which explains why it was silent on boot despite the volume being set via pot.