EdgeTX/edgetx #4548: Custom music doesn't play.

inventor7777, 2024-01-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On my RadioMaster MT12, I have placed custom WAV sound files (all shorter than 10 seconds each). But they don't play...at all. I even went to Settings-SD Card-Sounds-EN-held down-pressed play. The stock sounds work, but not my custom files. 

### Expected Behavior

I expected the files to play like the standard sounds.

### Steps To Reproduce

Add .wav files to SD card-sounds-en

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Operating System (OS)

macOS

### Anything else?

I'm using the nightly 2.10.0 (came with RM MT12)

The sound files are WAV files.

pfeerick, 2024-01-16:
What encoding? You can't use just any wave file... it should be 32khz PCM,
not sure if mono was a requirement but certainly no benefit as 99% of
radios (inc. MT12) are mono.

3djc, 2024-01-16:
Audio requirement:
Sample Rate: 32 kHz (or 16 Khz, 8Khz)
Bits / Sample: 16 (or 8)
Tracks / Channels: 1, mono
Compression Codec: PCM (or u-law, a-law)

pfeerick, 2024-01-16:
u-law and a-law are no longer valid. Only PCM.

inventor7777, 2024-01-16:
So would these settings be correct, or did I misunderstand something? (I still can't get them to work)

@pfeerick @3djc 

[image]

3djc, 2024-01-16:
Clearly not, your sample rate is wrong

TramALot, 2024-01-16:
If you look somewhere in the wiki, it explains everything and I believe it
recommends a program called audacity? I'm not sure of the spelling, But
it's free and will encode in the whatever You need as per it or above.

Helipro, 2024-01-18:
> So would these settings be correct, or did I misunderstand something? (I still can't get them to work)
> 
> @pfeerick @3djc
> 
> ![Screenshot 2024-01-16 at 13 10 12](https://private-user-images.githubusercontent.com/142270525/297163784-124231f9-31e3-4d53-8646-7d7cc4c8ca75.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU2MDcyMjQsIm5iZiI6MTcwNTYwNjkyNCwicGF0aCI6Ii8xNDIyNzA1MjUvMjk3MTYzNzg0LTEyNDIzMWY5LTMxZTMtNGQ1My04NjQ2LTdkN2NjNGM4Y2E3NS5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDExOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMThUMTk0MjA0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWI2MTk0MDJkMDY3NGNhOTI1ODY0MTEyYzRiYmVhYTUyOTAzMDhhYjhlMDExZmQ3MWFjNzJhZWI5MGI0MDI0MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.XNOLGAlNMHfYbLvTYC6esI7zk9qaYKgccfhSoZGdPIk)
Use Audacity.  Import your sound file.  I use notevibes.com
Change project rate to 32000hz
Export audio sound file in WAV 16-bit PCM
Done and your sound files will play.

closed 2024-01-23 by inventor7777