EdgeTX/edgetx #4556: only 5 labels are displayed simultaneously

ParkerEde, 2024-01-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

in the model manager, only 5 labels are displayed simultaneously in the main. In 2.9 there were still 6 labels. The spacing of the lines has become higher in Main. 

### Expected Behavior

I would very much appreciate it if the label line height could be set so that 6 labels can be seen at the same time again.

### Steps To Reproduce

Create 6 labels and observ modelmanager in 2.9 and main

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS), RadioMaster TX16S / TX16SMK2

philmoz, 2024-01-24:
UI element size for touch screen is 32 pixels + a few pixels padding.
The list box height was 30 pixels; but was changed to 35 for touch consistency.

ParkerEde, 2024-01-24:
And that's the problem, with X10 etc without touch you don't need touch consistency. I personally think more information on the screen is more valuable.

philmoz, 2024-01-24:
> And that's the problem, with X10 etc without touch you don't need touch consistency. I personally think more information on the screen is more valuable.

In an ideal world that would be great.

The UI code is already complex (and messy) having to support landscape and portrait orientations and touch vs non-touch interactions. Adding yet another layout scenario with compressed size/spacing would make development and testing even harder. With limited time and resources it's not practical right now.

gagarinlg, 2024-01-24:
It is relatively simple:
FrSky does not only not support EdgeTX, they oppose the existence of EdgeTX 
We do not have unlimited development resources

-> all FrSky radios have the state "community support", which basically means that someone has to take care of them. We don't even own or have access to all FrSky radios and variant between the core dev team.
When there are technical issues we will do our best to fix those, but special handling of those radios regarding UI design and similar things is some that will not happen unless a developer comes forward and takes the responsibility to test that radio specifically to make the necessary code changes.

I am sure that this is not what you wanted to hear and I am sorry for that, but ee have to draw lines somewhere.

ParkerEde, 2024-01-25:
Thank you for your reply. What I still don't quite understand is why this has been changed in general? Before (2.9) it also worked perfectly and was clearly touchable on the TX16S.
So be it, I'll close here.

closed 2024-01-25 by ParkerEde

pfeerick, 2024-01-25:
As I think I mentioned in another issue when you mentioned this before, it changed when all controls of the same type where made consistent in size. The "labels sidebar" is actually the same list control used in the popup/Choice menus. So when they got made a bit bigger, so did the labels. 

IMO, this is a touch vs non-touch issue, not anything to do with frsky/x10 - i.e. Jumper T16 would have the same issue. But, since the X10 (and T16 for that matter) can be modded to have a touch screen, and I don't believe there is a runtime mechanism in place to determine whether to be more touch or hard-key oriented, little can be done about this. As always, code contributions to improve the state of affairs are always welcome.