EdgeTX/edgetx #4562: M100 GPS Support

smcStevesmc, 2024-01-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have 2 good, know to work, HFLRC M100 GPS units and neither one will work in my RadioMaster RX16S MK II on EdgeTX

### Describe the solution you'd like

Working GPS

### Describe alternatives you've considered

Different FIrmware
Different OS
Giving up

pfeerick, 2024-01-25:
You probably need to reconfigure those GPSs... they need to be outputting NMEA messages, and it seems they are configured for UBOX binary by default. 

https://www.hglrc.com/products/m100-5883-gps

> 6. Output protocol: UBLOX

rotorman, 2024-04-21:
Have a look at: https://github.com/EdgeTX/edgetx/pull/4689 (merged to main).

closed 2025-02-14 by pfeerick