EdgeTX/edgetx #4569: Color: wrap around widget screens

ParkerEde, 2024-01-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With the Page <, Page > buttons on a TX16S you can scroll through the widget screen page by page. When I have reached the last page and continue scrolling, a wrap around takes place and I am back on the first page. With touch radios such as the TX16S, the screens can also be scrolled through using a touch swipe gesture. However, the wrap around does not work with the touch swipe gesture.

### Expected Behavior

Touch Swipe gesture should also work on the screen pages like the Page< and Page> buttons with wrap around

### Steps To Reproduce

create some screens.
Try to swipe through all screens
on the laste page try to swipe forward to come to the first page (wrap around)

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

current main

pfeerick, 2024-01-26:
As a counter to that... it could be argued the hard key implementation that allows the wrap-around is incorrect... since once you are at the end, you are at the end... and it should stop. Either/or.

philmoz, 2024-01-26:
The main view page swipe gesture for scrolling is handled by LVGL. Wrap around is not supported.

ParkerEde, 2024-01-26:
the hard key implementation with the wrap around is very useful and allows very good operation. It would be a big step backwards to remove this.
If it can't be done with the touch swipe gesture, that's okay, but the hard key implementation shouldn't change.

closed 2024-01-26 by ParkerEde